smartdevicelink/sdl_core #2887: Periodical core dump during low voltage

GetmanetsIrina, 2019-04-18:
### Bug Report
Periodical core dump during low voltage

##### Occurrences
Sometimes

##### Reproduction Steps
Issue is reproduced with script https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/SDL5_0/LowVoltage/001_Low_Voltage_and_Wake_Up.lua

##### Expected Behavior
Test script execution result is passed.

##### Observed Behavior
Test script execution result is failed, core dump is observed.

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/commit/12c02a9406277a43c43b0b069a12595534f6f7b5
* ATF: https://github.com/smartdevicelink/sdl_atf/commit/b99f18d369321eb1b1f6041022f9cd9748ec9242

##### Logs
[SDL_adn_ATF_logs.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3092956/SDL_adn_ATF_logs.tar.gz)

jacobkeeler, 2021-05-21:
@LitvinenkoIra @dboltovskyi Are you still able to reproduce this? I haven't yet been able to, though I know it is an occasional crash

dboltovskyi, 2021-05-26:
@jacobkeeler Confirmed. Issue not reproduced at https://github.com/smartdevicelink/sdl_core/tree/develop (84b0746527) so can be closed

closed 2021-05-26 by dboltovskyi