smartdevicelink/sdl_core #2888: Button Press notifications are not received on device when SEEK_LEFT hard button pressed when App in HMI Level LIMITED

LitvinenkoIra, 2019-04-18:
### Bug Report
#### Button Press notifications not received on device when SEEK_LEFT hard button pressed when App in HMI State LIMITED

##### Reproduction Steps 
1 .SDL and HMI are started
2. Media app is registered, activated and subscribed to the "SEEK_LEFT" button
3. Change the app level to LIMITED
4. HMI sends OnButtonPress ("SEEK_LEFT", appID, buttonPressMode)

##### Expected Behavior
SDL forwards the OnButtonPress notification to the mobile app

##### Observed Behavior
SDL  doesn't forward the OnButtonPress notification to the mobile app

-----------------------------------------------------------------------------------------

Also during the investigation of this problem I faced with issue that: **HMI API is mismatch with the current implementation** 
According to the current implementation OnButtonPress and OnButtonEvent notifications will be transferred to the subscribed app in **LIMITED** HMI level only if the **notification from the HMI contains the appID param**. 

Which is mismatch with the description in HMI API :
 https://github.com/smartdevicelink/sdl_core/blob/develop/src/components/interfaces/HMI_API.xml#L3853
https://github.com/smartdevicelink/sdl_core/blob/develop/src/components/interfaces/HMI_API.xml#L3870

Initially this logic for sending the notification to the app in LIMITED was applicable only for the "OK" button. But it was changed within:
- https://github.com/smartdevicelink/sdl_core/pull/576  
- https://github.com/smartdevicelink/sdl_core/pull/952 
- https://github.com/smartdevicelink/sdl_core/pull/1044 
- https://github.com/smartdevicelink/sdl_core/pull/775

So for now we have the following behavior:

- _If appID is present for **ANY BUTTON**  -> SDL transfers notification to the named app only if it is in FULL or LIMITED (ignores if app is in NONE or BACKGROUND)._
- _If appID is omitted for **ANY BUTTON** (except CUSTOM button) -> SDL transfers notification to app in FULL_

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: [release/5.0.1](https://github.com/smartdevicelink/sdl_core/blob/release/5.0.1)
* Testing Against: [release/5.0.1](https://github.com/smartdevicelink/sdl_core/blob/release/5.0.1) (applicable for the  [release/5.1.0](https://github.com/smartdevicelink/sdl_core/blob/release/5.1.0) also)

Logs and ATF test script are attached: 
[ATF_OnButtonPress.zip](https://github.com/smartdevicelink/sdl_core/files/3095140/ATF_OnButtonPress.zip)

E-SAITO-TMC, 2020-05-13:
@theresalech -san, We were able to reproduce this issue with Core 6.0.1. This is a high priority issue for Toyota. Could you please include this in the Core 6.2 release planning?

theresalech, 2020-05-21:
@LitvinenkoIra can you please advise if you are able to reproduce this issue in the latest Core release?

dboltovskyi, 2020-06-11:
@theresalech Please notice we're not planning to create the fix in scope of 6.2, though will try to do as much as possible

iCollin, 2020-07-17:
@dboltovskyi I see the attached ATF test completely passing against current develop. Is the issue of button presses not being transferred to apps in LIMITED still actual?

dboltovskyi, 2020-07-17:
@iCollin Confirmed. Issue is not reproduced on a current develop (5e2dff6d31)

iCollin, 2020-07-17:
@LitvinenkoIra I have submitted a fix for the subscribed apps in LIMITED issue in #3461, please let me know if this does fix the reported issue. Thank you.

iCollin, 2020-07-23:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3461 and https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2434

PR #3461 (2020-07-23): allow button events to be forwarded to apps in LIMITED
Fixes #2888

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
ATF test to be submitted (modified test 2888 with no appID param except for OK button)

### Summary
Will allow `OnButtonEvent` and `OnButtonPress` to be sent to mobile apps subscribed to that soft button while in LIMITED HMI level for all buttons besides `OK`.

Edit HMI_API description of `OnButtonEvent` and `OnButtonPress` param `appID` to remove inaccurate line:
```
In case the ButtonName is CUSTOM_BUTTON or OK, HMI must include ap [... cut]
closed 2020-07-23 by iCollin