smartdevicelink/sdl_core #2905: Add Code Analysis tools scripts

AByzhynar, 2019-05-02:
### Summary
Add the following tools for code quality checks and potential 
problems discovering on early stages: 
-  **cppcheck** -   is an analysis tool for C/C++ code. It detects the types of bugs that the compilers normally fail to detect. The goal is no false positives.
-  **cpplint** - Static code checker for C++
-  **pylint** -  code analyzer for Python (needed as lots of Python scripts are used in project)

**Note**: the tools specified above should be combined into one script and included into pre-commit hooks to make checks automatic

zcloud-jzy, 2019-05-05:
good

jacobkeeler, 2019-05-08:
Note: `cppcheck` is already included in the Travis CI build (https://github.com/smartdevicelink/sdl_core/blob/master/.travis.yml#L23)

theresalech, 2020-10-21:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3485.

closed 2020-10-21 by theresalech