smartdevicelink/sdl_core #2909: SDL does not put "count_of_rejected_rpc_calls" counter into PTS

aderiabin, 2019-05-08:
### Bug Report
SDL does not put "count_of_rejected_rpc_calls" counter into Policy Table Snapshot during Policy table update sequence

##### Reproduction Steps
0. Application data for App1 (appId = 0000001) exists in PT 
  RPC AddCommand is disallowed for App1  PT
  Value of counter "count_of_rejected_rpc_calls" is 0 for App1 is 0 in PT
  
1. SDL and HMI are started
2. Mobile №1 is connected to SDL and is consented
3. Application App1 is registered on Mobile №1
4. Application App1 from Mobile №1 is activated and sends valid AddCommand RPC request to SDL
5. Trigger PTU (for example: Register new application)

##### Expected Behavior
Value of `policy_table.usage_and_error_counts.app_level.0000001.count_of_rejected_rpc_calls` is 1 in PTS

##### Observed Behavior
Value of `policy_table.usage_and_error_counts.app_level.0000001.count_of_rejected_rpc_calls` is 0 in PTS

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: built from commit 94a46dd9cdbb7b8a281c067f81c188f66cfc6a5d

##### Test Case, Sample Code, and / or Example App
ATF script to reproduce issue is attached.

aderiabin, 2019-05-08:
The same issue applicable to other counters too:
`count_of_rejections_nickname_mismatch`
`count_of_rpcs_sent_in_hmi_none`
`count_of_removals_for_bad_behavior`
`count_of_user_selections`
etc.

jacobkeeler, 2021-05-21:
@aderiabin @dboltovskyi are you still able to reproduce this issue? I'm seeing these parameters in the snapshot with the correct values

dboltovskyi, 2021-05-26:
@jacobkeeler Confirmed. Issue not reproduced at https://github.com/smartdevicelink/sdl_core/tree/develop (84b0746527) so can be closed

closed 2021-05-26 by dboltovskyi