smartdevicelink/sdl_core #2910: SDL doesn't send OnSystemRequest to Mobile App during retry sequence

dboltovskyi, 2019-05-08:
### Bug Report
In case if Mobile App doesn't provide PT data during PTU within defined timeout SDL starts PTU retry sequence/
SDL should send OnSystemRequest to Mobile App several times during this process until sequence is finished.

##### Reproduction Steps
1. SDL is built in default policy mode (PROPRIETARY)
2. Register a new app => PTU is triggered and SDL sends `OnSystemRequest` to mobile app
3. Wait for some time until PTU retry sequence is started
4. Check OnSystemRequest is not send to mobile app

##### Expected Behavior
SDL sends OnSystemRequest to mobile app on every try

##### Observed Behavior
SDL does not send OnSystemRequest to mobile app during PTU retry sequence

##### OS & Version Information
* OS/Version: Ubuntu 16.04 amd64
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/release/5.1.0
* ATF: https://github.com/smartdevicelink/sdl_atf/tree/develop

##### Test Case, Sample Code, and / or Example App
[ATF_script.lua.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3158924/ATF_script.lua.tar.gz)
[2019-05-08_15-05-22_FAILED.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3158927/2019-05-08_15-05-22_FAILED.tar.gz) (with fix in SDL)
[2019-05-08_15-06-04_PASSED.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3158925/2019-05-08_15-06-04_PASSED.tar.gz) (SDL 5.1.0 release)

dboltovskyi, 2019-05-08:
It's very likely the root cause of the issue is the same as in https://github.com/smartdevicelink/sdl_core/issues/1210

jacobkeeler, 2020-04-20:
Closing with the merge of #3329

closed 2020-04-20 by jacobkeeler