smartdevicelink/sdl_core #2919: [SDL 0221] Remote Control - Allow Multiple Modules per Module Type

theresalech, 2019-05-21:
Proposal: [Remote Control - Allow Multiple Modules per Module Type](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0221-multiple-modules.md)

> SDL remote control baseline proposal [SDL-0071](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0071-remote-control-baseline.md) groups available controllable items of the same category into a RC module type. SDL defines six module types: `RADIO`, `CLIMATE`, `AUDIO`, `LIGHT`, `HMI_SETTINGS` and `SEAT`. SDL allows only one module per module type so far. However, there are production needs to support multiple modules of the same module type. For example, some vehicles in the market are equipped with "dual-zone" climate control systems. There are also ["tri-zone" or even "quad-zone" climate control](https://www.autotrader.com/car-info/definitions-dual-zone-climate-control-211783) in the existing vehicles. Some vehicles in the market have seat control for several seats (driver's seat, front passenger's seat for example). Some vehicles have both a default entertainment system and a back seat entertainment system. There are car headrest monitors, seat back monitors, car roof mount monitors and flip down screens. Passengers can use a back seat entertainment system to play video, audio or even games.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/700

Steering Committee Decision:

>The Steering Committee voted to accept this proposal with the following revisions:

- The default `userLocation` should be DRIVER to ensure backwards compatibility
- Add new non-mandatory `userLocation` parameter to `SetGlobalProperties` RPC instead of including `userLocation` as a parameter in each Remote Control request, such as `GetInteriorVehicleDataConsent`, SetInteriorVehicleData`, etc.

> The proposal .md file was updated to reflect these revisions on 5/21/19.

theresalech, 2019-08-01:
Proposal markdown file has been updated per the revisions included in accepted Evolution Proposal [SDL 0221 Revisions - Remote Control - Allow Multiple Modules per Module Type](https://github.com/smartdevicelink/sdl_evolution/issues/791).  Please reference issue and markdown file for revision details.

jordynmackool, 2019-09-06:
Closed by #2984

closed 2019-09-06 by jordynmackool