smartdevicelink/sdl_core #2922: SDL crashes while sending OnSystemRequest in case if RPC service is protected

dboltovskyi, 2019-05-22:
### Bug Report
SDL crashes while sending OnSystemRequest in case if RPC service is protected

##### Reproduction Steps
1. Register new app
=> Policy Update is started
2. Switch RPC service to protected mode
=> SDL continues PTU sequence
=> SDL->App: OnSystemRequest(with snapshot)

##### Expected Behavior
SDL proceed with OnSystemRequest notification successfully

##### Observed Behavior
SDL crashes with error:
```
smartDeviceLinkCore: ./sdl_core/src/components/security_manager/src/ssl_context_impl.cc:478: 
virtual bool security_manager::CryptoManagerImpl::SSLContextImpl::Encrypt(const uint8_t*, size_t, const uint8_t**, size_t*): 
Assertion `(len == static_cast<size_t>(read_size))' failed.
```

##### OS & Version Information
* OS/Version: Ubuntu 16.04 amd64
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/release/5.1.1
* ATF: https://github.com/smartdevicelink/sdl_atf/tree/develop
* Issue in SmartDeviceLink: TBA

##### Test Case, Sample Code, and / or Example App
[001_Protected_Mode_PTU.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3209075/001_Protected_Mode_PTU.tar.gz)
[2019-05-22_12-28-31_PROPRIETARY.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3209077/2019-05-22_12-28-31_PROPRIETARY.tar.gz)
[2019-05-22_12-28-51_EXTERNAL_PROPRIETARY.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3209078/2019-05-22_12-28-51_EXTERNAL_PROPRIETARY.tar.gz)

dboltovskyi, 2019-08-29:
@theresalech Please triage this issue. 
It's very likely it could affect https://github.com/smartdevicelink/sdl_core/issues/2816 which has been merged recently.

theresalech, 2019-09-03:
@dboltovskyi can you please confirm if you're able to reproduce this issue on `develop`?

dboltovskyi, 2019-09-03:
@theresalech Confirmed. Issue is reproduced on a current develop (e7b5c75). 
Attached script was slightly updated due to changes introduced by `Read Generic Signal Data` feature.

theresalech, 2019-09-04:
@dboltovskyi okay thank you.  We will investigate.  Do you anticipate submitting a fix for this issue as well?

dboltovskyi, 2019-09-04:
@theresalech Yes, we're going to submit a fix

dboltovskyi, 2019-09-19:
@theresalech Please notice PRs for `core` and `scripts` are ready for review

iCollin, 2019-09-19:
Closed with merge of #3021

PR #3021 (2019-09-19): Fix crash on secure rpc service
Fixes #2922 

This PR is **[ready]** for review.

### Risk
This PR makes **[no]** API changes.

### Testing Plan
ATF scripts

### Summary
When maximum allowed block size was calculated, maximum DTLS MTU was not taken into account, hence `BIO_read` resulted in error, which in turn lead to  failing `DCHECK` .

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2019-09-19 by iCollin