smartdevicelink/sdl_core #2929: Build fails on latest develop

LitvinenkoIra, 2019-05-24:
### Step to reproduce:
1. Check out latest develop branch (commit afc34d149d3b2eebab2df93e4b6783ad7716658f)
2. create a build directory ($ cd .. && mkdir cmake_build && cd cmake_build)
3. Give a build ($ cmake ../sdl_core && make)

### Expected Result:
SDL Core should build without any issue

### Actual Result:
Build fails with following error message:

```
12:30:48 [ 25%] Performing update step for 'libbson'
12:30:49 Current branch master is up to date.
12:30:49 No stash found.
12:30:49 No stash found.
12:30:49 No stash found.
12:30:49 CMake Error at /home/jenkins/workspace/develop_nightly/build/src/3rd_party/libbson-prefix/tmp/libbson-gitupdate.cmake:136 (message):
12:30:49   
12:30:49 
12:30:49   Failed to unstash changes in:
12:30:49   '/home/jenkins/workspace/develop_nightly/build/src/3rd_party/bson_c_lib/'.
12:30:49 
12:30:49 
12:30:49   You will have to resolve the conflicts manually
12:30:49 
12:30:49 
12:30:49 src/3rd_party/CMakeFiles/libbson.dir/build.make:94: recipe for target 'src/3rd_party/libbson-prefix/src/libbson-stamp/libbson-update' failed
12:30:49 make[6]: *** [src/3rd_party/libbson-prefix/src/libbson-stamp/libbson-update] Error 1
```

### Tested environment:
Ubuntu 16.04 (x86_64)

jacobkeeler, 2019-05-29:
This PR should have fixed the issue, but it isn't in develop yet because it was being included in the next hotfix: https://github.com/smartdevicelink/sdl_core/pull/2928. Please verify that this fixes the issue on your end.

LitvinenkoIra, 2019-05-31:
@jacobkeeler Thanks.
Verified on the latest develop https://github.com/smartdevicelink/sdl_core/commit/2026979753640570e5a6bfe239106a90f7c6ffdc
SDL Core builds without any issue

closed 2019-05-31 by LitvinenkoIra