smartdevicelink/sdl_core #2935: SDL destroys currently running mobile app requests on app unregistration

AKalinich-Luxoft, 2019-05-30:
### Bug Report
In the current SDL code base there is an actual issue related to command destruction at the moment when EventDispatcher executes `on_event()` in another thread.

Technically, at the moment when application unregisters, `RequestController::terminateWaitingForResponseAppRequests` function is invoked which erases all the commands from `waiting_for_response_` list which is holding the last shared pointer to the pending commands. The problem is that another thread may invoke `on_event` method for one of commands which are in `waiting_for_response_`. By that reason, `on_event` might be potentially invoked for a destroyed command what may cause a core crash.

##### Reproduction Steps
1. Register any application over BT transport
2. Send any request from registered application (e.g. Alert)
3. Disconnect transport at the moment when HMI potentially sends a response to SDL OR send UnregisterAppInterface from mobile

Note that issue is related to timings and it is hard to reproduce.

##### Expected Behavior
SDL unregisters application and drops the command when it is executed

##### Observed Behavior
Sometimes SDL crashes while trying to drop the command

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: 12c02a9406277a43c43b0b069a12595534f6f7b5

E-SAITO-TMC, 2020-04-17:
@joeygrover -san @theresalech -san This issue is a top priority for Toyota. Does this occur with the reproduction steps described? Also, does this issue occur in the latest version?

theresalech, 2020-04-20:
Hi @E-SAITO-TMC, we have documented that this is priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.
@AKalinich-Luxoft, can you please advise if the Luxoft team is seeing this issue in Core 6.1?

dboltovskyi, 2020-05-11:
@theresalech Please notice issue is still actual for Core 6.1

theresalech, 2020-06-23:
@dboltovskyi @AKalinich-Luxoft, can you please advise if Luxoft intends to submit a PR for this issue?  If so, can you please provide approximate timing?  Thanks!

dboltovskyi, 2020-06-23:
@theresalech Please notice we suppose this issue is going to be fixed by https://github.com/smartdevicelink/sdl_core/pull/3439 (fix for https://github.com/smartdevicelink/sdl_core/issues/3141 issue) which is in progress now. Our plan is to complete this fix in scope of `6.2` release.

theresalech, 2020-06-24:
@dboltovskyi, thank you!  Please let us know when PR 3439 is ready for Livio review.

theresalech, 2020-07-28:
@dboltovskyi, it appears PR #3439 has been closed.  Can you please advise if Luxoft will be submitting another PR to resolve both this issue (2935) and #3141 for the 7.0 release?

dboltovskyi, 2020-07-29:
@theresalech Please notice PR https://github.com/smartdevicelink/sdl_core/pull/3439 was a draft. We're still planning to provide a fix for these issues.

theresalech, 2020-07-29:
@dboltovskyi, thanks!  Please let us know once the fix is ready for Livio review.

iCollin, 2020-11-06:
@dboltovskyi do you still plan to contribute a fix for this issue?

dboltovskyi, 2020-11-06:
@iCollin Yes, we've almost done with fix and going to publish it once internal review is finished.
We've faced a lot of issues while preparing the fix and that's why we haven't replied so long.

AKalinich-Luxoft, 2020-12-07:
@JackLivio @theresalech  this PR is ready for review
Please note that PR contains a lot of similar changes due to change in the class inheritance structure required for making a proper state synchronization for all request classes