smartdevicelink/sdl_core #2936: Unit test "app_services_commands_test_unity" FAILED on develop branch

MNikora, 2019-06-05:
### Bug Report
Unit test "app_services_commands_test_unity" FAILED on develop branch at Jenkins CI job. 
Artifacts and other details available by the link: [[Build #320]](http://opensdl-jenkins.luxoft.com:8080/view/develop/job/develop_nightly_coverage/320/)

Compressed log file attached: [LastTest.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3257419/LastTest.tar.gz)

##### Observed Behavior
`10:20:00 *** Error in `/home/jenkins/workspace/develop_nightly_coverage/build/src/components/application_manager/rpc_plugins/app_service_rpc_plugin/test/app_services_commands_test': corrupted size vs. prev_size: 0x0000000002358800 ***
10:20:00 ======= Backtrace: =========
10:20:00 /lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f159f2ad7e5]
10:20:00 /lib/x86_64-linux-gnu/libc.so.6(+0x7e9dc)[0x7f159f2b49dc]
10:20:00 /lib/x86_64-linux-gnu/libc.so.6(+0x81cde)[0x7f159f2b7cde]
10:20:00 /lib/x86_64-linux-gnu/libc.so.6(__libc_malloc+0x54)[0x7f159f2ba184]
10:20:00 /home/jenkins/workspace/develop_nightly_coverage/build/src/components/application_manager/rpc_plugins/sdl_rpc_plugin/libsdl_rpc_plugin.so(+0x20c029a)[0x7f15a35fb29a]
10:20:00 /home/jenkins/workspace/develop_nightly_coverage/build/src/components/application_manager/rpc_plugins/sdl_rpc_plugin/libsdl_rpc_plugin.so(+0x20c08d2)[0x7f15a35fb8d2]
10:20:00 /lib/x86_64-linux-gnu/libc.so.6(__cxa_finalize+0x9a)[0x7f159f27036a]
10:20:00 /home/jenkins/workspace/develop_nightly_coverage/build/src/components/application_manager/rpc_plugins/sdl_rpc_plugin/libsdl_rpc_plugin.so(+0x12bca03)[0x7f15a27f7a03]`

##### OS & Version Information
* OS/Version: Ubuntu 16.04 LTS (official Docker image)
* SDL Core Version: [sdl_core/develop](https://github.com/smartdevicelink/sdl_core/tree/develop) (Last commit [2026979](https://github.com/smartdevicelink/sdl_core/commit/2026979753640570e5a6bfe239106a90f7c6ffdc))

LuxoftAKutsan, 2019-06-05:
Issue can be observed not only during unit tests run. Issues also reproduced sometimes on SDL shutdown.

KhrystynaDubovyk, 2019-06-06:
@theresalech, could you please triage the issue.

theresalech, 2019-06-06:
@KhrystynaDubovyk we'll bring this before the Steering Committee to include in the next Core release.  Thanks!

iCollin, 2019-08-22:
Closed with merge of https://github.com/smartdevicelink/sdl_core/pull/2939

PR #2939 (2019-08-22): Fix shutdown crash static var colission
Fixes #2936

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
The problem in double destruction static variables allocated for dynamically linked libraries, shared libraries and binary.

 - Add -fvisibility=hidden to avoid collision of static variables. Remove from ABI all shared objects members except one that certanly needed. Mostly is a specific for gcc compiler

 - Add appropriate destruction of plugins
 - Add Delete exported function to each rpc plugin library. Delete funciton will destroy Plugin instance.
 - [... cut]
closed 2019-08-22 by iCollin