smartdevicelink/sdl_core #2966: SDL incorectly assigns default permissions for an app if they were defined in preloaded

dboltovskyi, 2019-07-18:
### Bug Report
SDL incorectly assigns default permissions for an app if they were defined in preloaded

##### Reproduction Steps
1. Update SDL preloaded file and add default permissions for some app. E.g.:
```
"app_policies": 
{
  "app":"default",
  "default": {
    ...
    "groups": ["Base-4"],
    ...
  },
  ...
}
```
2. Register app
3. Check `OnHMIStatus` and `OnPermissionsChange` are sent to app

##### Expected Behavior
`OnHMIStatus` and `OnPermissionsChange` notifications are sent to app
Since these are in `Base-4` func. group and allowed at `NONE` HMI level

##### Observed Behavior
`OnHMIStatus` and `OnPermissionsChange` notifications are **not** sent to app

##### OS & Version Information
* OS/Version: Ubuntu 16.04 amd64
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (18ca3b7)
* ATF: https://github.com/smartdevicelink/sdl_atf/tree/develop

##### Test Case, Sample Code, and / or Example App
[002_default_app_policies_issue.lua.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3407263/002_default_app_policies_issue.lua.tar.gz)
[2019-07-18_11-24-16.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3407264/2019-07-18_11-24-16.tar.gz)

GetmanetsIrina, 2020-02-20:
The small update was performed in the issue script. The archive with script is updated.

LitvinenkoIra, 2020-02-21:
@theresalech please triage the issue. SDL incorrectly assigns default permissions for an app if they were defined in preloaded, its reproduced on proprietary and external proprietary policy flows.
 We have already prepared the fix for this issue in https://github.com/smartdevicelink/sdl_core/pull/3244

theresalech, 2020-02-24:
@LitvinenkoIra this issue has not yet been raised to the Steering Committee as priority for Core 6.1. We will work to include if time allows, but would recommend raising as priority in the Steering Committee meeting if it's imperative the issue be included. Thanks!

iCollin, 2020-03-02:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3244

PR #3244 (2020-03-02): Add function UnwrapAppPolicies to CacheManager::Init() method
Fixes #2966

This PR is **[ready]** for review.

### Risk
This PR makes [no] API changes.

### Testing Plan
Covered within existing unit tests and ATF tests.

### Summary
This Pull Request is intended to fix issue with default permissions for applications, which have already presented in PT preloaded file. Main cause of this issue is absence of function UnwrapAppPolicies in CacheManager::Init() method, as result of which section with permissions for existed app wasn't mapped with default parameters (if PTU didn't happen, because ear [... cut]
closed 2020-03-02 by iCollin