smartdevicelink/sdl_core #2969: [SDL 0242] Subtle Alert Style

theresalech, 2019-07-22:
Proposal: [Subtle Alert Style](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0242-alert-style-subtle.md)

> This feature adds the `SubtleAlert` RPC, which presents a style of alert: the "subtle" alert. This is a "notification style" alert that does not take over the entire screen.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/777

Steering Committee Decision:

>The Steering Committee voted to accept this proposal with the following revision: update to use alternate approach 1, for the reasons identified in [this comment](https://github.com/smartdevicelink/sdl_evolution/issues/777#issuecomment-511516024) from the Project Maintainer.

> The proposal .md file was updated to reflect these revisions on 7/22/19.

theresalech, 2020-07-22:
Proposal markdown file has been updated per the revisions included in accepted Evolution Proposal [SDL 0242 Revisions - Subtle Alert Style](https://github.com/smartdevicelink/sdl_evolution/issues/1046).  Revisions include updating several RPC definitions, and adding a policy field to control rate limiting for `SubtleAlert` requests as is currently available for `Alert` requests.  Please reference the issue and proposal markdown file for revisions details.

theresalech, 2020-08-05:
Proposal markdown file has been updated per the revisions included in accepted Evolution Proposal [SDL 0242 Revisions - Subtle Alert Style](https://github.com/smartdevicelink/sdl_evolution/issues/1060).  Revisions include adding a `SubtleAlert`response, including all basic response parameters as well as `tryAgainTime`.  Please reference the issue and proposal markdown file for revisions details.

jacobkeeler, 2020-08-17:
Closing with the merge of #3459

closed 2020-08-17 by jacobkeeler