smartdevicelink/sdl_core #2970: [SDL 0238] Keyboard Enhancements

theresalech, 2019-07-22:
Proposal: [Keyboard Enhancements](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0238-Keyboard-Enhancements.md)

> This proposal is about adding a new keyboard layout and adding new properties for keyboards supported by SDL. The following keyboard enhancements are discussed in this proposal.

Enhancement #1: Support for Numeric Keyboard.

Enhancement #2: Allow apps to mask entered characters.

Enhancement #3: Allow apps to change special characters shown on the keyboard layout.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/754

Steering Committee Decision:

>The Steering Committee voted to accept this proposal with the revisions outlined in [this comment](https://github.com/smartdevicelink/sdl_evolution/issues/754#issuecomment-511379464) from the author.  The project maintainer also agreed to the author's point regarding `INVALID_DATA` (12).

> The proposal .md file was updated to reflect these revisions on 7/22/19.

jordynmackool, 2021-02-03:
The proposal markdown file has been updated per the revisions included in the accepted with revisions review issue: [Revise SDL-0238 Keyboard Enhancements](https://github.com/smartdevicelink/sdl_evolution/pull/1119).

Accepted revisions can be reviewed in the merged PR here: https://github.com/smartdevicelink/sdl_evolution/pull/1117.

jacobkeeler, 2021-02-12:
closing with the merge of #3609

closed 2021-02-12 by jacobkeeler