smartdevicelink/sdl_core #2973: SDL sends lockScreenDismissalEnabled=false in case LockScreenDismissalWarning is missing in Policy Table

dboltovskyi, 2019-07-25:
### Bug Report
SDL sends lockScreenDismissalEnabled=false in case LockScreenDismissalWarning is missing in Policy Table

##### Reproduction Steps
1. Make sure `LockScreenDismissalWarning` section is missing in PT
2. Start SDL, HMI
3. Register and activate app
4. HMI sends `OnDriverDistraction` notification twice: DD_OFF, DD_ON

##### Expected Behavior
SDL transfers notifications, and in case of DD_ON set lockScreenDismissalEnabled=true

##### Observed Behavior
SDL transfers notifications, and in case of DD_ON set lockScreenDismissalEnabled=false

##### OS & Version Information
* OS/Version: Ubuntu 16.04 amd64
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (b20c699)
* ATF: https://github.com/smartdevicelink/sdl_atf/tree/develop

##### Test Case, Sample Code, and / or Example App
[013_OnDD_PT_Warning_msg_absent_in_PT.lua.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3431300/013_OnDD_PT_Warning_msg_absent_in_PT.lua.tar.gz)
[2019-07-25_07-43-12.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3431301/2019-07-25_07-43-12.tar.gz)

IGapchuk, 2019-07-25:
The place of code, which causes the issue https://github.com/smartdevicelink/sdl_core/blob/b20c6990247a0bc88a6c3b86fd1d92530cd141e9/src/components/application_manager/rpc_plugins/sdl_rpc_plugin/src/commands/hmi/on_driver_distraction_notification.cc#L87

According to proposal:
`
In addition to modifying the notification, the proposed solution is to add a swipe down gesture to dismiss the lock screen (**as well as display the provided warning message**), similar to Android Auto. This gesture will be allowed if lockScreenDismissalEnabled is set to **true**, and disabled if set to **false**.
`

SDL shouldn't change `lockScreenDismissalEnabled` value in case of missing `LockScreenDismissalWarning` message.

MOBILE_API description of `lockScreenDismissalWarning` parameter:
 ```
Warning message to be displayed on the lock screen when dismissal is enabled.
This warning should be used to ensure that the user is not the driver of the vehicle, 
ex. `Swipe down to dismiss, acknowledging that you are not the driver.`.
This parameter must be present if "lockScreenDismissalEnabled" is set to true.
```
According to this description - `lockScreenDismissalEnabled` parameter doesn't depend on `lockScreenDismissalWarning`, but `lockScreenDismissalWarning` parameter depends on `lockScreenDismissalEnabled`

dboltovskyi, 2019-07-25:
@theresalech Please triage this issue

jacobkeeler, 2019-07-25:
By the description of `lockScreenDismissalWarning`, if no warning can be provided, `lockScreenDismissalEnabled` cannot be allowed. This is for safety reasons, we should not allow lock screen dismissal if no warning message is available. This should take precedence over the configuration in the policy table.

As the author of the `lockScreenDismissalWarning` revision, I can say that this was the intention behind this description. If we sent `lockScreenDismissalWarning=true` without `lockScreenDismissalWarning`, we would be going against this description.

jacobkeeler, 2019-07-25:
> According to proposal:
> In addition to modifying the notification, the proposed solution is to add a swipe down gesture to dismiss the lock screen (**as well as display the provided warning [...]

The description you are referencing is referring to the proxy behavior, which is dependent on the `lockScreenDismissalEnabled` value received from Core in `OnDriverDistraction`, not the actual policy configuration. It has no relevance to Core's behavior.

dboltovskyi, 2019-07-25:
@jacobkeeler Is there possible case when just icon (or picture) needs to be displayed instead of warning message?
In such one icon won't be displayed since `lockScreenDismissalEnabled=false`

jacobkeeler, 2019-07-25:
@dboltovskyi The SDLC would have to agree upon such an icon, and I personally don't think there is such an icon which would work. The important thing is that it communicates that the driver should never be the one dismissing the lock screen.

In any case, the current behavior is really just a failsafe. In practice, an OEM should ***always*** provide a warning message in the policy table if `lock_screen_dismissal_enabled=true`.

closed 2019-07-26 by jacobkeeler

dboltovskyi, 2019-07-26:
@jacobkeeler Agreed, if warning message is always provided this becames non-realistic case. 
Issue is closed.