smartdevicelink/sdl_core #2974: SDL processes the undefined in HMI specification info message from HMI response

GetmanetsIrina, 2019-07-26:
### Bug Report
SDL processes the undefined in HMI specification info message from HMI response

##### Preconditions
1. HMI and SDL are started
2. App is registered and activated

##### Reproduction Steps
1. Mobile app requests PerformInteraction(Manual only)with undefined message
2. HMI responds with structure result within choiceID and  info message with resultCode WARNINGS

##### Expected Behavior
According to HMI specification: 
```
//Response with no params in it (code and method are necessary)
{
	"id" : 36,
	"scorpion" : "2.0",
	"result" : 
	{
		"code" : 0,
		"method": "UI.Alert"
	}
}
```
result structure has no any messages

##### Observed Behavior
SDL processes info message in result struct from HMI and resend it to mobile app

##### OS & Version Information
* OS/Version:Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop(87a6f62)
* ATF: https://github.com/smartdevicelink/sdl_atf/tree/develop(d6c2138)
* Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/SDL5_0/Transfer_RPC_with_invalid_image/005_PerfomInteraction.lua

##### Logs
[ATF_and_SDL_log.zip](https://github.com/smartdevicelink/sdl_core/files/3436320/ATF_and_SDL_log.zip)

LitvinenkoIra, 2019-09-06:
SDL behaves according to the expected behavior - don't resend undefined 'info' parameter from a successful HMI response. ATF test script needs to be updated: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2255

iCollin, 2019-09-13:
Issue closed as problem actually resides in ATF.

closed 2019-09-13 by iCollin