smartdevicelink/sdl_core #2975: SDL sends empty SubscribeVehicleData request to HMI in case of resumption

GetmanetsIrina, 2019-07-26:
### Bug Report
SDL sends empty SubscribeVehicleData request to HMI in case of resumption

##### Preconditions
1. Mobile app is registered and has permissions for vehicle data_1
2. Mobile app subscribes to data_1
3. Ing off/unexpected disconnect is performed

##### Reproduction Steps
1. Ign on/transport connect is performed
2. App starts registration with actual hashId

##### Expected Behavior
SDL sends SubscribeVehicleData with data_1 to HMI during resumption

##### Observed Behavior
SDL sends empty SubscribeVehicleData HMI during resumption

##### OS & Version Information
* OS/Version:Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop(87a6f62)
* ATF: https://github.com/smartdevicelink/sdl_atf/tree/develop(d6c2138)
* Script: TBD

jacobkeeler, 2019-09-03:
@GetmanetsIrina can you verify if this was fixed in #2977

dboltovskyi, 2019-09-03:
@jacobkeeler Issue is fixed in scope of #2977 and can be closed.

closed 2019-09-23 by jacobkeeler