smartdevicelink/sdl_core #2976: SDL does not send OnSDLClose notification by switching off with reason IGNITION_OFF

aderiabin, 2019-07-29:
### Bug Report
SDL does not send OnSDLClose notification by switching off with reason IGNITION_OFF

#### Preconditions
1. SDL and HMI are started.
2. Mobile application is registered.

#### Reproduction Steps
1. Perform IGNITION_OFF (send BasicCommunication.OnExitAllApplications with reason = "IGNITION_OFF")

#### Expected Behavior
SDL sends OnSDLClose notification to HMI

#### Observed Behavior
SDL does not send OnSDLClose notification to HMI

#### Test Case, Sample Code, and / or Example App
Test script: 
https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Smoke/ShutDown/001_ShutDown_IGNITION_OFF.lua

aderiabin, 2019-08-01:
Test set [policies_happy_paths_PROPRIETARY](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_sets/policies_happy_paths_PROPRIETARY.txt) should be updated after closing this issue.

IGapchuk, 2019-09-02:
The defect doesn't reproduce.

**Environment:**
**OS:** Ubuntu 18.0
**SDL Core:** develop (https://github.com/smartdevicelink/sdl_core/commit/e7b5c75c704f6cd577f0857e4cbe17cf6ec160fa)
**ATF:** develop (https://github.com/smartdevicelink/sdl_atf/commit/d6c2138f8d894d1e343d971226a16c47019ee92e)

IGapchuk, 2019-09-06:
Currently the issue doesn't reproduce by switching off with reason IGNITION_OFF, but reproduces by reason MASTER_RESET.
Script for reproduce issue: https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Smoke/ShutDown/002_ShutDown_MASTER_RESET.lua

dboltovskyi, 2019-09-19:
@theresalech Please triage this issue

theresalech, 2019-09-19:
@dboltovskyi please advise if Luxoft will be submitting a PR to resolve this issue, and if so, when that PR is ready for Livio review.  Thanks!

dboltovskyi, 2019-09-19:
@theresalech Yes, Luxoft is preparing a fix now.

dboltovskyi, 2019-09-19:
@theresalech Fix is ready for review

theresalech, 2019-09-19:
@dboltovskyi got it, thanks!  We'll review https://github.com/smartdevicelink/sdl_core/pull/3022

PR #3022 (2019-09-19): Make usage logger stabilization
Fixes #2976

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
ATF script:
https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Smoke/ShutDown/001_ShutDown_IGNITION_OFF.lua

### Summary
Added custom deleter for logger
Added flush logger during each handle deleting
Added deleting logger thread during destroy each plugin
This changes are necessary to avoid deadlock during work with log4cxx.

### CLA
- [X] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-x [... cut]
closed 2019-09-19 by JackLivio