smartdevicelink/sdl_core #2983: SDL does not respond on incorrect requests from HMI

aderiabin, 2019-08-04:
### Bug Report
SDL does not respond on incorrect requests from HMI 

##### Reproduction Steps
0.Precondition:
  1) SDL and HMI are started
 1. Send incorrect RPC from HMI to SDL
Example: GetURLS(service = true) 

##### Expected Behavior
SDL sends response with error code `INVALID_DATA` to HMI

##### Observed Behavior
SDL does not respond on request from HMI

##### OS & Version Information
* OS/Version: Ubuntu 16.04 amd64
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (b7a3f9c)
* ATF: https://github.com/smartdevicelink/sdl_atf/tree/develop

##### Test Case, Sample Code, and / or Example App

E-SAITO-TMC, 2020-05-13:
@theresalech -san, We were able to reproduce this issue with Core 6.0.1. This is a medium priority issue for Toyota. Could you please include this in the Core 6.2 release planning?

theresalech, 2020-05-21:
@aderiabin can you please advise if this issue is reproducible in Core 6.1.1?