smartdevicelink/sdl_core #2988: SDL incorrect update interior vehicle data for LIGHT RC module with OnInteriorVehicleData notification

aderiabin, 2019-08-09:
### Bug Report
SDL incorrect update interior vehicle data for LIGHT RC module with OnInteriorVehicleData notification

##### Reproduction Steps
0.Precondition:
  1) SDL and HMI are started with App 1 and App 2 application data in LPT
  2) HMI sent LIGHT module capabilities to the SDL
  3) HMI has next data for LIGHT RC module:
```
```
  3) Mobile is connected to the SDL
  4) App 1 and App 2 are registered and activated

1. Send correct "GetInteriorVehicleData"(moduleType = "LIGHT", subscribe = true) request from App 1 to the SDL
2. Send correct "SetInteriorVehicleData" (moduleType = "HMI_SETTINGS", hmiSettingsControlData is correct) request from App2 to the SDL 

##### Expected Behavior
SDL sends "OnRCStatus" notifications to App 1 and App 2
SDL sends "RC.OnRCStatus" notifications to the HMI
SDL resends "RC.SetInteriorVehicleData" request to the HMI and receives respond from HMI
SDL responds with "SetInteriorVehicleData"(success = true, resultCode = "SUCCESS") to the App 2

##### Observed Behavior
SDL does NOT send "OnRCStatus" notifications to App 1 and App 2
SDL does NOT send "RC.OnRCStatus" notifications to the HMI
SDL does NOT resend "RC.SetInteriorVehicleData" request to the HMI
SDL responds with "SetInteriorVehicleData"(success = false, resultCode = "IN_USE") to the App 2

##### OS & Version Information
* OS/Version: Ubuntu 16.04 amd64
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (9a556df)
* ATF: https://github.com/smartdevicelink/sdl_atf/tree/develop

##### Test Case, Sample Code, and / or Example App
[test_script.zip](https://github.com/smartdevicelink/sdl_core/files/3451853/test_script.zip)
[logs.zip](https://github.com/smartdevicelink/sdl_core/files/3451859/logs.zip)

dboltovskyi, 2019-09-18:
@theresalech @jordynmackool Please notice PRs for `core` and `scripts` are ready for review

AStasiuk, 2019-10-02:
@theresalech, @jacobkeeler could you please close this issue, if no additional action required?

jacobkeeler, 2019-10-02:
Closing with the merge of #3017

PR #3017 (2019-09-20): Fix merge module data incorrect behavior
Fixes #2988 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Added unit tests for RCHelpers

### Summary
Moved MergeModuleData and MergeArray methods from InteriorDataCache to RCHelpers.
Previously, when data for module was saved into SDL cache, on update SDL reseted cached data with new one, no matter of data type. It is critical in case we have array of objects which have field _"id"_ .  

So instead of merging MergeArray function just replaced existing data with the new one.
(See detailed issue descript [... cut]
closed 2019-10-02 by jacobkeeler