smartdevicelink/sdl_core #2995: SDL sends UPDATE_NEEED even if app is not registered yet

dboltovskyi, 2019-08-22:
### Bug Report
SDL sends UPDATE_NEEED even if app is not registered yet

##### Reproduction Steps
1. Make sure SDL is built in EXTERNAL_PROPRIETARY policy mode
2. Start SDL, HMI, connect Mobile device
3. Consent Mobile device

##### Expected Behavior
SDL starts RPC service successfully and does not send `SDL.OnStatusUpdate(UPDATE_NEEED)` notification to HMI until app is successfully registered.

##### Observed Behavior
SDL starts RPC service successfully and sends `SDL.OnStatusUpdate(UPDATE_NEEED)` just after device is consented

##### OS & Version Information
* OS/Version: Ubuntu 16.04 amd64
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (c6e9a9b)
* ATF: https://github.com/smartdevicelink/sdl_atf/tree/develop
* Testing Against: ATF (script, report and logs are in attachment)

##### Test Case, Sample Code, and / or Example App
[2019-08-22_12-29-15.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3531054/2019-08-22_12-29-15.tar.gz)

dboltovskyi, 2019-10-18:
@theresalech Please triage this issue.
It's not a regression one, though we propose to include it in 6.0 release since fix is simple and it has successfully passed full regression cycle.

iCollin, 2019-10-21:
Closed via #3090

PR #3090 (2019-10-21): Check if there are any registered app before sending UPDATE_NEEDED
Fixes #2995

This PR is **[ready]** for review.

### Risk
This PR makes **[no]** API changes.

### Testing Plan
Policy ATF test scripts 

### Summary
Check if there are any registered app before sending UPDATE_NEEDED

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2019-10-21 by iCollin