smartdevicelink/sdl_core #2996: SDL doesn't WAKE_UP after LOW_VOLTAGE signal

dboltovskyi, 2019-08-23:
### Bug Report
SDL does not WAKE_UP after LOW_VOLTAGE signal

##### Reproduction Steps
1. Start SDL, HMI, connect Mobile, register App
2. Send `LOW_VOLTAGE` signal
3. Wait a few seconds and send `WAKE_UP` signal

##### Expected Behavior
SDL wakes up and starts resumption sequence

##### Observed Behavior
SDL doesn't wake up

##### OS & Version Information
* OS/Version: Ubuntu 16.04 amd64
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (8db0c26)
* ATF: https://github.com/smartdevicelink/sdl_atf/tree/develop
* Testing Against: ATF (script, report and logs are in attachment)

##### Test Case, Sample Code, and / or Example App
- [SDL5_0/low_voltage.txt](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/master/test_sets/SDL5_0/low_voltage.txt)
- [2019-08-23_10-17-50.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3534944/2019-08-23_10-17-50.tar.gz)

dboltovskyi, 2019-08-23:
This is regression issue which is introduced by fix for https://github.com/smartdevicelink/sdl_core/pull/2939

dboltovskyi, 2019-08-23:
@theresalech Please triage this issue

theresalech, 2019-09-25:
Closed via #3029

PR #3029 (2019-09-24): Fix SDL behavior during low voltage
Fixes #2996 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by ATF test scripts

### Summary
There were found a several problems during SDL testing with ATF scripts:
1. SDL ignores messages from mobile/HMI during low voltage but as they are still delivered to the independent external container (e.g. socket) during Low Voltage - when SDL wakes up - it is not aware what data was in the container and starts processing them. 
2. Sometimes SDL ignores messages on wake up due to timings
3. SDL low voltage pr [... cut]
closed 2019-09-25 by theresalech