smartdevicelink/sdl_core #2997: SDL does not apply the versioning for vehicle data

GetmanetsIrina, 2019-08-23:
### Bug Report
SDL does not apply the versioning for vehicle data

##### Preconditions
1. SDL and HMI are started
2. App has permissions for fuelRange, parameter is available `since` 5.0 version

##### Reproduction Steps
1. Register application with `majorVersion = 3, minorVersion = 0`
2. App requests `SubscribeVehicleData` with `fuelRange`

##### Expected Behavior
SDL procesess  the request with `fuelRange` as request with undefined data

##### Observed Behavior
SDL processed  the request with `fuelRange` as defined data

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/commit/8db0c2601921fdd368bf86133780746b532c40f5
ATF: https://github.com/smartdevicelink/sdl_atf/commit/d6c2138f8d894d1e343d971226a16c47019ee92e
Script: TBD

##### Logs
[report.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3535582/report.tar.gz)

iCollin, 2019-09-13:
Closed with merge of #3015

jacobkeeler, 2019-09-13:
This is not an actual issue, the reason that a base RPC version was introduced was because the proxies did not send an accurate RPC version before mobile API versioning was introduced. #3015 should be reverted

iCollin, 2019-09-13:
#3015 is being reverted by #3028

JackLivio, 2019-09-13:
I want to note that this is an existing issue and is not a regression cause by 6.0. We are marking it as low priority until all regression issues are resolved.

jacobkeeler, 2019-09-25:
@GetmanetsIrina I would argue this is a non-issue, as it was a goal of the Generic Network Signal Data feature for an older version of Core to be able to support future vehicle data items:

> OEM's schema file should always be baselined with latest version of standardized schema file from open SDL. OEMs can then further add OEM specific custom vehicle data items. This ensures that any version of core that implements this proposal would be able to process the latest version of standardized vehicle data item schema, and hence would be able to support all future additions for new standardized vehicle data items.

This behavior allows for this possibility. In addition, I don't think that it is an issue if an app requests data that isn't present in the negotiated message version. If an app is aware of the appropriate key for this data, there shouldn't be an issue with providing this data if it has the appropriate permissions.

GetmanetsIrina, 2019-09-26:
@jacobkeeler, if the app wants to request the `fuelRange` it should understand the response. The application is able to understand the response only in case when the `fuelRange` is available in `RPCSpec`. If application works with `majorVersion=3`, `fuelRange` is absent in `RPCSpec` that mobile app works with.
SDL should prevent the sending of the parameters that are absent in `RPCSpec` version, supported by mobile application. Except cases with OEM spec vehicle data, but OEM specific vehicle data schema should also follow the same versioning rules. 
For now SDL behaviour is not according to https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0089-mobile-api-versioning.md so this issue was created for updating of the SDL behaviour according to mobile versioning requirements.

jacobkeeler, 2019-09-26:
I would argue that if an app is requesting `fuelRange` despite registering as a `3.0.0` application, then it is definitely able to understand the response. As a note, the negotiated version of this application would be `4.5.1` due to a base RPC version being included, I've described this behavior in the [following comment](https://github.com/smartdevicelink/sdl_core/pull/3049#discussion_r328177862), though this would not affect this specific case. This is more a matter of strictness, and I am arguing that Core shouldn't be strict on this matter.

GetmanetsIrina, 2019-09-27:
@jacobkeeler, okay, got it and agree about the apps with version before 5.0, but are the same rules applicable for apps with version after 5.0(the case for example with app 5.0 and parameter since 6.0 )?

`The attribute since and until should be used for any element (enum, element, function, struct, param) in the mobile API to define the version range in that the element is valid` so in case the value has `since=6.0` it's not valid for the versions before `6.0`

jacobkeeler, 2019-09-27:
@GetmanetsIrina Alright, I'll admit that this behavior doesn't match up with the proposal as-is. I still would consider it a minor issue, but it is an issue.

One thing to note about developing a fix for this, it would need to account for possible newer vehicle data types in the policy table. Currently, the negotiated version of an app cannot go above the module version. For example:
1. An app attempts to register with version `7.0` in the future, the module version is `6.0`. As such, the negotiated version is 6.0
2. An app attempts to get a vehicle data type which was introduced in `6.1`, this data type is defined in the policy table and the app has permissions to it. This means that the app should be able to retrieve this data.

As such, Core will need to account for it's own max version, and allow any vehicle data parameters which were defined after this version

GetmanetsIrina, 2019-09-30:
@jacobkeeler, sorry, but i do not clearly understand, what do you mean `module version`( API version or max supported version or maybe anything else)?

jacobkeeler, 2019-09-30:
@GetmanetsIrina by module version I mean the [MOBILE_API version](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/interfaces/MOBILE_API.xml#L4) on Core's side (which would be the max supported version)

GetmanetsIrina, 2019-09-30:
@jacobkeeler, okay thanks, and in this case if the version of the MOBILE_API is 6.0 how can the API contain the value with 6.1 version?
I talk about the case in your example:
`
2. An app attempts to get a vehicle data type which was introduced in 6.1, this data type is defined in the policy table and the app has permissions to it. This means that the app should be able to retrieve this data
`

jacobkeeler, 2019-09-30:
> this data type is defined in the policy table and the app has permissions to it
This is possible because of #2840, this type is defined in the policy table, not in Core's local API

GetmanetsIrina, 2019-10-01:
@jacobkeeler , i want to be on the same page, so created the list of the cases with the expected result.
Could you please review it and approve in case everything meets the expectations.
[Cases_for_2997_vd_versioning.zip](https://github.com/smartdevicelink/sdl_core/files/3676404/Cases_for_2997_vd_versioning.zip)

jacobkeeler, 2019-10-01:
I'd say that these cases should be independent of the application version. 

Case 10 doesn't make sense:
> version of parameter in DB is less than app version but more than API version
because 
meaning that `API version < DB version < app version`, but due to version negotiation `app version <= API version`

I think that this could be reduced to 4 cases

| Case | Parameter type | Parameter version | Expected |
|------|-----------------|-------------------|-----------|
| 1    | parameter is listed only in DB    | version of parameter >= module version                                            | The app is able to retrieve parameter                                                                |
| 2    | parameter is listed only in DB    | version of parameter < module version                                            | The app is NOT able to retrieve parameter                                                                    |
| 3    | parameter is listed in API   | version of parameter > app version                                            | The app is NOT able to retrieve parameter                                                                |
| 4   | parameter is listed in API   | version of parameter <= app version                                            | The app is able to retrieve parameter                                                                    |

LuxoftAKutsan, 2019-10-03:
@jacobkeeler  OK, this table is a good tip for understanding SDL behavior. The only unclear point is the behavior of applications < 5.0. 
Am I right that SDL should not apply any versioning rules if the application version < 5.0 ?  (points 1, 2, 3) in @GetmanetsIrina  table.

jacobkeeler, 2019-10-03:
@LuxoftAKutsan versioning is still applied to pre-5.0, but the application's version for API versioning purposes will be assigned to a base version of 4.5.1

Ex. If an application sends an RAI with a syncMsgVersion of `3.0.0` to a module with Core 5.0+, it will be assigned the version `4.5.1` instead of the version that was sent.

YarikMamykin, 2019-11-22:
@theresalech please notice that PR https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2256 is ready for re-review.

GetmanetsIrina, 2019-11-25:
@theresalech, please notice that PR with tests https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2256 is also ready for review.

jordynmackool, 2019-11-25:
@GetmanetsIrina thanks, we will review and let you know if we have any questions.

jacobkeeler, 2020-01-15:
Closing with the merge of #3049

PR #3015 (2019-09-13): Fix sdl versioning for vehicle data
Fixes #2997 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Added unit tests for RPCHandlerImpl class (GetMessageVersion method)
Adjust ATF scripts for GetVehicleData

### Summary
Corrected processing of SyncMsgVersion according to mobile application version (on RegisterAppInterface).
Corrected validating of incoming parameters from mobile request.

### Tasks Remaining:
- [ ] [Adjust ATF scripts for GetVehicleData]

### CLA
- [X] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482 [... cut]
PR #3049 (2020-01-15): Fix sdl versioning for vehicle data
Fixes #2997 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Added unit tests for RPCHandlerImpl class (GetMessageVersion method)
Adjust ATF test scripts for GetVehicleData.
Fix existing ATF tests for GenericNetworkSignalData, Resumption.

### Summary
Corrected processing of SyncMsgVersion according to mobile application version (on RegisterAppInterface).
Corrected validating of incoming parameters from mobile request.

### CLA
- [X] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX [... cut]
closed 2020-01-15 by jacobkeeler