smartdevicelink/sdl_core #3002: Build break on latest develop branch when ENABLE_LOG is OFF

dboltovskyi, 2019-08-30:
### Bug Report
Build break on latest develop branch when ENABLE_LOG is OFF

##### Reproduction Steps
1. Try to build SDL with cmake ` -ENABLE_LOG=OFF` flag

##### Expected Behavior
SDL is built successfully

##### Observed Behavior
Build failed

##### OS & Version Information
* OS/Version: Ubuntu 16.04 amd64
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (0186b8c)

##### Test Case, Sample Code, and / or Example App
- Console log
```
[  7%] Building CXX object src/components/protocol_handler/CMakeFiles/ProtocolHandler.dir/src/handshake_handler.cc.o
In file included from /home/db/ramdrv/sdl_core/src/components/include/utils/message_queue.h:39:0,
                 from /home/db/ramdrv/sdl_core/src/components/protocol_handler/include/protocol_handler/protocol_handler_impl.h:42,
                 from /home/db/ramdrv/sdl_core/src/components/protocol_handler/include/protocol_handler/handshake_handler.h:38,
                 from /home/db/ramdrv/sdl_core/src/components/protocol_handler/src/handshake_handler.cc:33:
/home/db/ramdrv/sdl_core/src/components/include/utils/conditional_variable.h:36:47: fatal error: boost/thread/condition_variable.hpp: No such file or directory
 #include <boost/thread/condition_variable.hpp>
```

dboltovskyi, 2019-08-30:
Please note this is regression issue which has been introduced by implementation of #2840

dboltovskyi, 2019-08-30:
@theresalech Please triage this issue

theresalech, 2019-09-03:
@dboltovskyi can you please advise if you will be submitting a fix for this issue?

dboltovskyi, 2019-09-03:
@theresalech Yes, we're going to prepare a fix.

jacobkeeler, 2019-09-04:
Closing with the merge of #3013

PR #3013 (2019-09-04): Fix build with disabled logger.
Fixes #3002

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
Since boost and logger include are mostly located in the same directory, there is no issue building with the logger.
In `stl_utils.h` NULL is not defined while building `stl_utils_test`.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2019-09-04 by jacobkeeler