smartdevicelink/sdl_core #3003: PTU trigger on ignition cycles doesn't work properly

dboltovskyi, 2019-09-01:
### Bug Report
When the amount of ignition cycles notified by HMI via `BasicCommunication.OnIgnitionCycleOver` gets equal to the value of
-- `exchange_after_x_ignition_cycles` field (`module_config` section) of policies database, SDL must trigger a PTU sequence.

Currently SDL sometimes doesn't increment it's internal counter of ignition cycles

##### Reproduction Steps
1. Make sure `exchange_after_x_ignition_cycles` in local PT = 5
1. Start SDL, HMI, connect Mobile, register App
2. Perform 5 ignition cycles

##### Expected Behavior
SDL starts PTU on a 5th ignition cycle

##### Observed Behavior
SDL doesn't start PTU on a 5th ignition cycle

##### OS & Version Information
* OS/Version: Ubuntu 16.04 amd64
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (e7b5c75)
* ATF: https://github.com/smartdevicelink/sdl_atf/tree/develop
* Testing Against: ATF (script, report and logs are in attachment)

##### Test Case, Sample Code, and / or Example App
- [build_options/109_ATF_PTU_Trigger_IGN_cycles_PROPRIETARY](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Policies/build_options/109_ATF_PTU_Trigger_IGN_cycles_PROPRIETARY.lua)
- [2019-09-01_01-00-01.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3563110/2019-09-01_01-00-01.tar.gz)

dboltovskyi, 2019-09-01:
Please notice this is regression issue was introduced by implementation of https://github.com/smartdevicelink/sdl_core/issues/2840

jacobkeeler, 2019-09-03:
Closing with the merge of #3005

PR #3005 (2019-09-03): Add check for already stored vehicle data item params
Fixes #3003

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
During vehicle data item parameter storage to DB, the new (and the same) row will be inserted to parent-parameter relation table for each one's version. As a result, during the retrieve of a composite vehicle data item, such parameter could be retrieved several times. And after several cycles, parameter array will exceed max boundary and therefore will be invalid, so the storage process fails and some data (including meta as `exchange_afte [... cut]
closed 2019-09-03 by jacobkeeler