smartdevicelink/sdl_core #3014: SDL does not  resend Error GetVehicleData response from HMI

VjKlepikov, 2019-09-04:
### Bug Report
SDL does not  resend Error GetVehicleData response from HMI

##### Reproduction Steps
1. Start SDL, HMI, connect Mobile, register new App
2. Send GetVehicleData with speed = true
3. Send HMI response with {"error":{"code":9,"data":{"method":"VehicleInfo.GetVehicleData"}},"id":57,"jsonrpc":"2.0"}

##### Expected Behavior
1) SDL sends to HMI VehicleInfo.GetVehicleData: "params":  { "speed":true }
2) SDL sends to mobile response of GetVehicleData with {"resultCode":"VEHICLE_DATA_NOT_AVAILABLE", "success":false, "speed":50.5}

##### Observed Behavior
1) SDL sends to HMI VehicleInfo.GetVehicleData: "params":  { "speed":true }
2) SDL sends to mobile response of GetVehicleData with {"resultCode":"GENERIC_ERROR","success":false}

##### OS & Version Information
* OS/Version: OS/Version: Ubuntu 16.04 amd64
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (bc8288d)
* ATF: https://github.com/smartdevicelink/sdl_atf/tree/feature/script_runner (e4b9574)

##### Test Case, Sample Code, and / or Example App
[logs.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3575404/logs.tar.gz)
[script.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3575407/script.tar.gz)

dboltovskyi, 2019-09-04:
Please notice this is regression issue was introduced by implementation of https://github.com/smartdevicelink/sdl_core/issues/2840

dboltovskyi, 2019-09-04:
@theresalech Please triage this issue

theresalech, 2019-09-04:
@dboltovskyi please advise when a fix has been submitted and we will review.

dboltovskyi, 2019-09-04:
@theresalech Sure, our team is working on it now. I suppose it will be ready in a day or two.

iCollin, 2019-09-13:
Closed with merge of #3016

PR #3016 (2019-09-13): Error message processing from HMI
Fixes #3014, #2439

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
Fix for error message processing received from HMI (#3014).
Also, there is a fix for invalid result code received on Mobile (#2439). (previous fix is not actual due to new features merged into develop)

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2019-09-13 by iCollin