smartdevicelink/sdl_core #3026: SDL sends OnStatusUpdate(UPDATE_NEEDED) for existing app

dboltovskyi, 2019-09-13:
### Bug Report
SDL sends OnStatusUpdate(UPDATE_NEEDED) for existing app

##### Reproduction Steps
1. Start SDL, HMI, connect Mobile, register App
2. Perform successful PTU => `UP_TO_DATE`
3. Re-connect app

##### Expected Behavior
SDL doesn't send `OnStatusUpdate(UPDATE_NEEDED)` since app already has it's policies and previous status was `UP_TO_DATE`

##### Observed Behavior
SDL sends OnStatusUpdate(UPDATE_NEEDED)

##### OS & Version Information
* OS/Version: Ubuntu 16.04 amd64
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (2449d7e)
* ATF: https://github.com/smartdevicelink/sdl_atf/tree/develop
* Testing Against: ATF (script, report and logs are in attachment)

##### Test Case, Sample Code, and / or Example App
Failed scripts:
 - [SSLHandshakeFlow/015_Navi-Predefined_cert_valid_PTU_is_not_started](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Security/SSLHandshakeFlow/015_Navi-Predefined_cert_valid_PTU_is_not_started.lua)
 - [SSLHandshakeFlow/016_Non-Navi-Predefined_cert_valid_PTU_is_not_started](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Security/SSLHandshakeFlow/016_Non-Navi-Predefined_cert_valid_PTU_is_not_started.lua)
 - [Policies/build_options/109_ATF_PTU_Trigger_IGN_cycles_PROPRIETARY](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Policies/build_options/109_ATF_PTU_Trigger_IGN_cycles_PROPRIETARY.lua)

dboltovskyi, 2019-09-13:
This is regression issue which is introduced by implementation of fix for https://github.com/smartdevicelink/sdl_core/issues/3019

dboltovskyi, 2019-09-13:
@theresalech @jordynmackool Please triage this issue

jordynmackool, 2019-09-13:
@mked-luxoft will do, please let us know once the PR is ready for Livio review!

iCollin, 2019-09-23:
Closed with merge of #3035

PR #3035 (2019-09-23): Add condition check for policy ResetRetrySequence
Fixes #3026 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Manual testing described in #3026 

### Summary
Fixes an issue with #3020 which made it so `UPDATE_NEEDED` was sent whenever all apps were disconnected, this fix changes the behavior of `StopRetrySequence` to only reset the sequence if it is currently in progress.

### Changelog
##### Bug Fixes
* Changes the behavior of `StopRetrySequence` to only reset the sequence if it is currently in progress.

### CLA
- [x] I have signed [the CLA](h [... cut]
closed 2019-09-23 by iCollin