smartdevicelink/sdl_core #3027: SDL cuts off custom parameters for App Services

dboltovskyi, 2019-09-13:
### Bug Report
Within App Services it was introduced a possibility to define custom parameters for service data.
SDL cuts off all these parameters in `PublishAppService` response

##### Reproduction Steps
1. Start SDL, HMI, connect Mobile, register App
2. App publishes app service with custom parameters in manifest
3. SDL processed with request successfully 

##### Expected Behavior
SDL provides custom parameter in response to mobile

##### Observed Behavior
SDL cuts off custom parameter

##### OS & Version Information
* OS/Version: Ubuntu 16.04 amd64
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (73b3483)
* ATF: https://github.com/smartdevicelink/sdl_atf/tree/develop
* Testing Against: ATF (script, report and logs are in attachment)

##### Test Case, Sample Code, and / or Example App
 - [AppServices/GetAppServiceData/004_M2M_unknown_param_success](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/AppServices/GetAppServiceData/004_M2M_unknown_param_success.lua)
 - [AppServices/GetAppServiceData/007_M2M_Future_service_type_success_flow](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/AppServices/GetAppServiceData/007_M2M_Future_service_type_success_flow.lua)
 - [AppServices/GetAppServiceData/017_H2M_MediaImage_success](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/AppServices/GetAppServiceData/017_H2M_MediaImage_success.lua)

dboltovskyi, 2019-09-13:
This is regression issue which is introduced by implementation of fix for #2997

dboltovskyi, 2019-09-13:
@theresalech @jordynmackool Please triage this issue

iCollin, 2019-09-13:
@dboltovskyi Thank you for noticing so quickly.

This issue is resolved with the merge of #3028 (which reverts #3015 )

closed 2019-09-13 by iCollin