smartdevicelink/sdl_core #3031: SDL does not clear all user consent records in "user_consent_records" section in case of FACTORY_DEFAULTS

dboltovskyi, 2019-09-17:
### Bug Report
Sometimes SDL does not clear all user consent records in "user_consent_records" section of the LocalPT after `FACTORY_DEFAULTS`

##### Reproduction Steps
0. Make sure SDL is built in `EXTERNAL_PROPRIETARY` mode
1. Start SDL, HMI, connect Mobile, register App
2. Provide consent for the device and functional group
3. Perform Factory reset by sending `BC.OnExitAllApplications(FACTORY_DEFAULTS)` from HMI
4. Start SDL

##### Expected Behavior
There is no data in ‘device_consent_group and ‘consent_group’ tables

##### Observed Behavior
Data in ‘device_consent_group and ‘consent_group’ tables is present

##### OS & Version Information
* OS/Version: Ubuntu 16.04 amd64
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (3ea6b1a)
* Testing Against: ATF (script, report and logs are in attachment)

##### Test Case, Sample Code, and / or Example App
 - Script: [Policies/user_consent_of_Policies/221_ATF_Factory_reset](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Policies/user_consent_of_Policies/221_ATF_Factory_reset.lua)

dboltovskyi, 2019-09-17:
There is a closed issue https://github.com/smartdevicelink/sdl_core/issues/1225 which describes the same steps. However the root cause of current issue is different.
Current issue may be connected to https://github.com/smartdevicelink/sdl_core/issues/2976

dboltovskyi, 2019-09-19:
@theresalech Please triage this issue

theresalech, 2019-09-19:
@dboltovskyi please advise if Luxoft will be submitting a PR to resolve this issue, and if so, when that PR is ready for Livio review. Thanks!

dboltovskyi, 2019-09-19:
@theresalech Yes, we're preparing a fix now

dboltovskyi, 2019-09-25:
@theresalech Please notice fix is ready for PM review.

PR #3048 (2019-09-26): Fix/crash on ign off
Fixes #3031 
This PR is **[ready]** for review.

### Risk
This PR makes **[no]** API changes.

### Testing Plan
ATF tests

### Summary
`RPCServiceImpl` has two message queues, that were not stopped during SDL shutdown procedure. This resulted in one the queues trying to handle a message when other components had been already deleted. To fix that, a `Stop` procedure was introduced to `RPCServiceImpl`. 

### CLA
- [ ] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2019-09-26 by JackLivio