smartdevicelink/sdl_core #3041: Need help in sdl core build on Android Platform

amitchaturvedi, 2019-09-19:
Working on a critical assignment where we need SDL core on ANdroid Platform.
Can you provide the necessary steps  to compile the same. It is very urgent since we want to integrate this ASAP.

Regards,
AMIT
Senior Architect

jordynmackool, 2019-09-20:
Hi @amitchaturvedi, 

Please use the available documentation on our site to further assist you. If you have specific questions, please post them in the appropriate slack channels (http://slack.smartdevicelink.com) and someone from our team or from the SDL community will respond. 
 
- https://smartdevicelink.com/en/guides/hmi/overview/
- https://smartdevicelink.com/en/guides/core/overview/
- https://smartdevicelink.com/en/guides/android/getting-started/installation/

closed 2019-09-20 by jordynmackool

amitchaturvedi, 2019-09-20:
Hi,
SDL core can only run on Linux platform as per content available on SDL website .
 "SDL core" support is not available for Android platform. 

I have a HU running on Android. I want to run SDL core on Android platform. Can you please let me know if we can run this SDL core on Android and what are the steps to build SDL core with Android platform

jordynmackool, 2019-09-20:
@amitchaturvedi Please post your questions in slack for assistance. The sdl_core repo should only be used when reporting an issue or submitting a PR related to core.

amitchaturvedi, 2019-09-20:
Provide the link for slack please..I will post my query there ..

jordynmackool, 2019-09-20:
http://slack.smartdevicelink.com