smartdevicelink/sdl_core #3042: SDL rejects requests for custom VihicleData

dboltovskyi, 2019-09-19:
### Bug Report
In [SDL-0173 Read Generic Network Signal data](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0173-Read-Generic-Network-Signal-data.md) it was introduced a possibility for SDL to handle custom `VehicleData` parameters.
These parameters can be defined through standard PTU process.
**Sometimes** SDL is unable to proceed with just set custom parameters.  

##### Reproduction Steps
1. Start SDL, HMI, connect Mobile, register App
2. Perform PTU with setting of new custom `VehicleData` parameters
3. Send `SubscribeVehicleData` with subscription to one (or more) new parameter(s)

##### Expected Behavior
SDL proceed with subscription successfully

##### Observed Behavior
SDL rejects request with `INVALID_DATA` result code

##### OS & Version Information
* OS/Version: Ubuntu 16.04 amd64
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (3ea6b1a)
* ATF: https://github.com/smartdevicelink/sdl_atf/tree/develop
* Testing Against: ATF (script, report and logs are in attachment)

##### Test Case, Sample Code, and / or Example App
 - Script: [GenericNetworkSignalData/001_SubscribeVehicleData_with_PTU_with_VDitems](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/API/VehicleData/GenericNetworkSignalData/001_SubscribeVehicleData_with_PTU_with_VDitems.lua)
 - Logs:

AKalinich-Luxoft, 2019-09-19:
Seems like SDL sends `UP_TO_DATE` status before all internal SDL data actually updated. According to attached SDL logs:

1. SDL sends UP_TO_DATE at 13:58:20:500
```
DEBUG [2019-09-19 13:58:20,500][0x7f6c2a7fc700][RPCServiceImpl] /home/jenkins/workspace/Regression_experimental_SDL_NoUT_External_Proprietary/src/components/application_manager/src/rpc_service_impl.cc:761 rpc_service::RPCServiceImpl::ConvertSOtoMessage: Conversion result: {
   "jsonrpc" : "2.0",
   "method" : "SDL.OnStatusUpdate",
   "params" : {
      "status" : "UP_TO_DATE"
   }
}
```

2. Execution of callback for `CustomVehicleDataManagerImpl` to update all custom parameters by some reason takes almost 1 second:
```
TRACE [2019-09-19 13:58:20,521][0x7f6c2a7fc700][VehicleInfoPlugin] /home/jenkins/workspace/Regression_experimental_SDL_NoUT_External_Proprietary/src/components/application_manager/rpc_plugins/vehicle_info_plugin/src/custom_vehicle_data_manager_impl.cc:475 CustomVehicleDataManagerImpl::OnPolicyEvent: Enter
...
TRACE [2019-09-19 13:58:21,129][0x7f6c2a7fc700][VehicleInfoPlugin] /home/jenkins/workspace/Regression_experimental_SDL_NoUT_External_Proprietary/src/components/application_manager/rpc_plugins/vehicle_info_plugin/src/custom_vehicle_data_manager_impl.cc:475 CustomVehicleDataManagerImpl::OnPolicyEvent: Exit
```

3. SDL receives SubscribeVD from ATF during `OnPolicyEvent` and cuts it off as a fake param:
```
DEBUG [2019-09-19 13:58:20,797][0x7f6c2affd700][RPCHandlerImpl] /home/jenkins/workspace/Regression_experimental_SDL_NoUT_External_Proprietary/src/components/application_manager/src/rpc_handler_impl.cc:329 rpc_handler::RPCHandlerImpl::ConvertMessageToSO: 			Message to convert: protocol 2; json {"custom_vd_item4_string":true}
...
DEBUG [2019-09-19 13:58:20,798][0x7f6c2affd700][ApplicationManager] /home/jenkins/workspace/Regression_experimental_SDL_NoUT_External_Proprietary/src/components/application_manager/src/message_helper/message_helper.cc:3215 MessageHelper::PrintSmartObject: SMART OBJECT: {
   "msg_params" : {},
   "params" : {
      "connection_key" : 65537,
      "correlation_id" : 4,
      "function_id" : 20,
      "message_type" : 0,
      "protection" : false,
      "protocol_type" : 0,
      "protocol_version" : 2
   }
}
```

Note, that ATF expects that SDL finishes all required PTU actions when it receives `UP_TO_DATE` from SDL, however this is not true for now.

dboltovskyi, 2019-09-20:
@theresalech Please triage this issue. 
Our team is preparing a fix.

jacobkeeler, 2019-09-20:
@AKalinich-Luxoft So the issue is caused just because Core isn't able to update the vehicle data schemas quickly enough, causing the ATF script to fail? I can't see this issue being particularly relevant in a production case, as it would require an app to request this data immediately after a PTU in which this data was added, which seems unlikely. Marking this as low priority for now

AKalinich-Luxoft, 2019-09-23:
@jacobkeeler yes, you are right. It's unlikely for a production case so will work on it with the lower priority

jacobkeeler, 2019-10-01:
Closing with the merge of #3051

PR #3051 (2019-10-01): Fix PTU notifications order
Fixes #3042 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by ATF test scripts

### Summary
The existing order of actions were changed to make it more obvious for external systems like
HMI and mobile apps. The main concern was that SDL is sending `OnStatusUpdate(UP_TO_DATE)` when policy table update was not actually done. The order of actions done during PTU was updated
to perform sending of `UP_TO_DATE` only when all policy actions were actually done.

### Tasks Remaining:
- [x] [Fix failed unit tests]
- [x] [... cut]
closed 2019-10-01 by jacobkeeler