smartdevicelink/sdl_core #3053: Sometimes SDL crashes when unpublishing app service

dboltovskyi, 2019-09-26:
### Bug Report
Sometimes SDL crashes which unpublishing app service 
Quite often (up to 50% runs) issue is happen with [002_GetSystemCapabilitySubscribe_OnSystemCapability_on_unregister](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/AppServices/SystemCapabilities/002_GetSystemCapabilitySubscribe_OnSystemCapability_on_unregister.lua) ATF script on `HTTP` policy flow

##### Reproduction Steps
1. Start SDL, HMI, connect Mobile, register App
2. App publishes some app service
3. App sends `UnregisterAppInterface`

##### Expected Behavior
SDL correctly unpublish the service

##### Observed Behavior
SDL crashed

##### OS & Version Information
* OS/Version: Ubuntu 16.04 amd64
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (35e6a7e)
* ATF: https://github.com/smartdevicelink/sdl_atf/tree/develop
* Testing Against: ATF (script, report and logs are below)

##### Test Case, Sample Code, and / or Example App
 - Script: [002_GetSystemCapabilitySubscribe_OnSystemCapability_on_unregister](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/AppServices/SystemCapabilities/002_GetSystemCapabilitySubscribe_OnSystemCapability_on_unregister.lua)
 - Report: [2019-09-26_17-27-24.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3659986/2019-09-26_17-27-24.tar.gz)
 - Stack trace: [stack_trace_correct.txt](https://github.com/smartdevicelink/sdl_core/files/3659968/stack_trace_correct.txt)

dboltovskyi, 2019-09-26:
@theresalech Please triage this issue.
Please notice our team already prepared a fix which is in testing now.

jacobkeeler, 2019-09-27:
Closing with the merge of #3053

PR #3056 (2019-09-27): Fix SDL crash on app service unpublishing
Fixes #3053 

This PR is **not ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by ATF test script

### Summary
There was found an unsafe work with STL map in `AppServiceManager` - during service unpublishing there might be a possibility when component iterates over `published_services_` and another function indirectly erases the same element from the same map what may cause an undefined behavior during iteration over the map.
To make this safer - iteration over the map and erasure of element in the map were spl [... cut]
closed 2019-09-27 by jacobkeeler