smartdevicelink/sdl_core #3059: SDL is unable to start protected Audio/Video service in multi-transport configuration

dboltovskyi, 2019-10-01:
### Bug Report
Within https://github.com/smartdevicelink/sdl_core/pull/2164 it was introduced a possibility for SDL to handle multiple transports. E.g.: RPC service could utilize BT connection whereas Audio/Video services use WIFI.
Currently SDL is unable to start secure Audio/Video service in such configuration even if RPC service in protected mode has been started successfully.

##### Reproduction Steps
1. Set up SDL for secure sessions:
   - set SDL certificate  
   - set Mobile root and issuing CA certificates
2. Set up SDL for multiple transports:
   - MultipleTransportsEnabled = true
   - SecondaryTransportForBluetooth = WiFi
   - AudioServiceTransports = TCP_WIFI
   - VideoServiceTransports = TCP_WIFI
3. Set force protection for Audio/Video services:
   - ForceProtectedService = 0x0A, 0x0B
4. Make sure SDL and Android mobile device connected to the same WIFI and BT
5. Start SDL, Web-HMI
6. In SPT select `IPC MP` connection type (which means multiple transports)
7. Register new NAVI app
8. Switch RPC service to protected mode (this step is optional; it's added only to be sure secure connection can be established for RPC service)
9. Activate app on Web-HMI
10. Try to start secure Video streaming

##### Expected Behavior
SDL starts secure streaming successfully

##### Observed Behavior
SDL responds with NACK to mobile app

##### OS & Version Information
* OS/Version: Ubuntu 18.04 amd64
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (66888d4)
* Testing Against: FordSPT, Web-HMI

##### Test Case, Sample Code, and / or Example App
 - [SmartDeviceLinkCore_NACK_encrypted.log.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3677830/SmartDeviceLinkCore_NACK_encrypted.log.tar.gz)

dboltovskyi, 2019-10-02:
@theresalech Please triage this issue.
It seems It's not a regression one, though it's highly appreciate to be a part of 6.0 release.
Our team is working on a fix now.

theresalech, 2019-10-02:
@dboltovskyi please let us know when a fix is ready for review and we'll do our best to include in the release.  We're currently focused on preparing the initial Release Candidate for Steering Committee review (delivering 2019-10-04), so may need to consider this issue after the initial RC is delivered.

dboltovskyi, 2019-10-04:
@theresalech Please notice fix has been passed our testing and ready for review

PR #3065 (2019-10-09): Fix SSL handshake for a secondary connection
Fixes #3059 

This PR is **not ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Tested manually

### Summary
There was found a several places in SDL code where SSL handshake was working wrong when secured service was started from a secondary connection like WIFI. All these places were updated to consider primary connection as a source for an SSL context and other attributes for case when the current connection is secondary.
Looks like this is a part which was not considered during initial implementation of the second [... cut]
closed 2019-10-09 by iCollin