smartdevicelink/sdl_core #3061: SDL doesn't send Navigation.OnAudioDataStreaming when app stops streaming

dboltovskyi, 2019-10-02:
### Bug Report
SDL doesn't send `Navigation.OnAudioDataStreaming` notification to HMI when app stops streaming

##### Reproduction Steps
1. Start SDL, HMI, connect Mobile, register NAVI app
2. App starts streaming
=> SDL sends `Navigation.OnAudioDataStreaming(available = true)` to HMI
3. App stops streaming

##### Expected Behavior
SDL sends `Navigation.OnAudioDataStreaming(available = false)` to HMI

##### Observed Behavior
SDL doesn't send notification to HMI

##### OS & Version Information
* OS/Version: Ubuntu 16.04 amd64
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (7c442ab)
* Testing Against: ATF (script, report and logs are in attachment)

##### Test Case, Sample Code, and / or Example App
  - Script: [003_Start_audio_streaming](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/MobileProjection/Phase1/003_Start_audio_streaming.lua)
  - Report and logs: [2019-10-02_11-05-58.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3681839/2019-10-02_11-05-58.tar.gz)

dboltovskyi, 2019-10-02:
@theresalech Please triage this issue.
Please notice this is regression issue which is introduced by fix for #2945

dboltovskyi, 2019-10-02:
Please notice, there are more scripts failed due to this issue:
 - [006_Start_audio_service_streaming_in_LIMITED](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/MobileProjection/Phase1/006_Start_audio_service_streaming_in_LIMITED.lua)
 - [017_HappyPath_flow_with_audio_video_streamings](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/MobileProjection/Phase1/017_HappyPath_flow_with_audio_video_streamings.lua)

iCollin, 2019-10-03:
@dboltovskyi Thank you for reporting!

Closed via https://github.com/smartdevicelink/sdl_core/pull/3064

PR #3064 (2019-10-03): ensure correct order of stream length calculations
Fixes #3061

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
ATF

### Summary
Change where `socket_audio_stream_start_time_` is initialized. 
It is possible that `MediaManagerImpl::OnMessageReceived` is called with data before `MediaManagerImpl::StartStreaming` is called, causing calculations for how long the stream will be to fail because `socket_audio_stream_start_time_` was not initialized.

### Changelog
##### Bug Fixes
* Fix bug of stream length miscalculations with an uninitialized `socket_a [... cut]
closed 2019-10-03 by iCollin