smartdevicelink/sdl_core #3073: SDL does not set the app to NONE HMI level after a rejecting of StartService

GetmanetsIrina, 2019-10-11:
### Bug Report
SDL does not set the app to NONE HMI level after a rejecting of StartService

##### Precondition
1. HMI and SDL are started
2. Navigation app is registered and activated

##### Reproduction Steps
1. App requests start of the protected video/audio/rpc service
2. Protected service can't be started by any reason: invalid cert, invalid ptu, not successful response to GetSystemTime

##### Expected Behavior
1. SDL responds with NACK to mobile application
2. SDL sends BC.OnServiceUpdate(serviceEvent = "REQUEST_REJECTED") to HMI with the appropriate reason 
3. SDL sets app to NONE HMI level and sends OnHMILevel(NONE) to mobile application 

##### Observed Behavior
SDL does not sent app to NONE HMI level and does not send the OnHMILevel(NONE) to mobile application 

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/commit/2b09ea3adf962e25717b6b4cd751173f0aa40d02
* HMI: https://github.com/smartdevicelink/sdl_hmi/tree/develop(1db5de1dfb5afbbbcdca266662faeb0d7d2050e4)
* SPT: https://rink.hockeyapp.net/apps/dbc7db5274027027262185753824e34d
* Scripts: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2280

Logs:
[2019-10-11_12-09-16.zip](https://github.com/smartdevicelink/sdl_core/files/3716733/2019-10-11_12-09-16.zip)

GetmanetsIrina, 2019-10-11:
@theresalech, Please triage this issue.

theresalech, 2019-10-14:
@GetmanetsIrina can you please advise if Luxoft anticipates submitting a PR to fix this issue?

dboltovskyi, 2019-10-14:
@theresalech Yes, our team is going to prepare a fix.

theresalech, 2019-10-14:
@dboltovskyi thank you!  Please let us know when it is ready for Livio review.

dboltovskyi, 2019-10-17:
@theresalech Please notice both fixes in `core` and `scripts` are ready for review

JackLivio, 2019-10-25:
Fixed by #3082

PR #3082 (2019-10-25): Fix SDL does not set the app to NONE HMI level after a rejecting of StartService
Fixes #3073

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
There is the issue when SDL tries to start streaming service (VIDEO or AUDIO) and receives a rejected request from HMI (e.g. GetSystemTime(INVALID_TIME)). In this case, SDL should send to mobile device OnHMIStatus("NONE") notification and send CloseApplication request to HMI.
In the current implementation in the case when SDL receives a rejected request, it sends just OnServiceUpdate notification to HMI with reje [... cut]
closed 2019-10-25 by JackLivio