smartdevicelink/sdl_core #3075: TextFieldName enum mis-matched between APIs

iCollin, 2019-10-14:
## Bug Report
Currently the `HMI_API.xml`'s `TextFieldName` enum has two elements that do not exist in the `MOBILE_API.xml`'s `TextFieldName` enum. These elements are `notificationText` and `navigationText`.

This problem was exposed by issue #1392 and temporarily mitigated by PR #3074 .

##### Expected Behavior
An enum defined in both `HMI_API.xml` and `MOBILE_API.xml` should match.

##### Observed Behavior
An enum defined in both `HMI_API.xml` and `MOBILE_API.xml` has additional elements in the `HMI_API.xml`.

##### OS & Version Information
* SDL Core Version: 6.0 RC1

LitvinenkoIra, 2020-04-23:
Hello, @iCollin I have a couple of questions related to this issue.

 As I can see there is a mismatch with **three elements** in the `TextFieldName` enum:  `"navigationText", "notificationText", "timeToDestination"`
I can see these elements in the [HMI_API](https://github.com/smartdevicelink/sdl_core/blob/release/6.1.0/src/components/interfaces/HMI_API.xml) but they do not exist in the [Mobile_API](https://github.com/smartdevicelink/rpc_spec/blob/master/MOBILE_API.xml)
 _btw I was not able to found 6.1.0 tag in the rps_spec, therefore, I put the link to the master branch_

1. Could you please advise if the PM plans to submit a PR to resolve this issue or Luxoft can fix it? 
2. Could you please clarify - these elements should be added to the Mobile_API or should be removed from the HMI_API?

Thank you.

VjKlepikov, 2020-05-07:
@iCollin @LitvinenkoIra Please note that [HMI_API](https://github.com/smartdevicelink/sdl_core/blob/release/6.1.0/src/components/interfaces/HMI_API.xml) has four params at "TextFieldName" enum omitted at [MOBILE_API](https://github.com/smartdevicelink/rpc_spec/blob/master/MOBILE_API.xml)
  <element name="timeToDestination"/> -> omitted at MOBILE_API
    
  <element name="turnText"/> -> omitted at MOBILE_API
  <element name="navigationText"> -> omitted at MOBILE_API
    <description>Navigation text for UpdateTurnList.</description>
  </element>
  <element name="notificationText"> -> omitted at MOBILE_API
    <description>Text of notification to be displayed on screen.</description>
  </element>

iCollin, 2020-05-07:
Hello @LitvinenkoIra ,

Yes it seems there are four mismatched elements in this enum.

The root cause is this commit to prepare the spec for release 4.0: https://github.com/smartdevicelink/sdl_core/commit/a91f49224cf94614f1753433ad96473fd4572207

1. PM does plan to resolve this issue.
2. All elements are still used by core excluding `notificationText`. The solution will be to remove `notificationText` from the HMI API and add the other three names to the MOBILE API. This will require a proposal to implement so all SDLC members are made aware of these changes. I plan to create the proposal in the coming days.

iCollin, 2020-05-07:
@LitvinenkoIra the proposal has now been created: https://github.com/smartdevicelink/sdl_evolution/pull/1008

jordynmackool, 2021-09-13:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3595

closed 2021-09-13 by jordynmackool