smartdevicelink/sdl_core #3076: SDL applies PT file received when PTU procedure is not in progress

aderiabin, 2019-10-15:
### Bug Report
SDL applies PT file received when PTU procedure is not in progress

##### Precondition
1. Policy table contains permissions for application (App) 
   `GetVehicleData` RPC  is not allowed for App.
2. HMI and SDL are started
3.  App is registered and activated (PTU is not triggered)

##### Reproduction Steps
1. App sends valid "SystemRequest" with PT file which allows `GetVehicleData` RPC for App
2. App sends  valid `GetVehicleData` RPC

##### Expected Behavior
SDL responds on `GetVehicleData` with DISALLOWED resultCode

##### Observed Behavior
SDL responds on `GetVehicleData` with SUCCESS resultCode 

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/releases/tag/6.0.0_RC ( 2b09ea3 )
* HMI: https://github.com/smartdevicelink/sdl_hmi/tree/develop ( 1db5de1 ) 
* SPT: SyncProxyTester ( 20191001 Nightly-Android ) 

Logs:
[hmi.log](https://github.com/smartdevicelink/sdl_core/files/3728748/hmi.log)
[SmartDeviceLinkCore.log](https://github.com/smartdevicelink/sdl_core/files/3728749/SmartDeviceLinkCore.log)

aderiabin, 2019-10-15:
This is not a regression issue and reproduced on a previous SDL release (5.1.3)

aderiabin, 2019-10-16:
@theresalech, Please triage this issue. 
Luxoft team is going to prepare a fix.

theresalech, 2019-10-16:
@aderiabin we've noted that this is not a regression issue but will work to prioritize reviewing the PR once submitted.  Thanks!

JackLivio, 2019-10-18:
I agree that there is an issue present. Allowing an unsolicited PTU SystemRequest from an app could raise some security concerns if there is no PTU encryption in place.

We could update Core to keep track of the appID it sent the onSystemRequest to and only accept the PTU SystemRequest from that app id. Its not a 100% fix because a bad actor could still receive that first OnSystemRequest.

Even though I agree with this issue, I still want to submit an evolution proposal for a fix to ensure SystemRequests are only accepted from apps during PTU in progress state. The proposal will be for visibility to other SDLC members in case someone's policy flow was defined around being able to send unsolicited PTUs.

LitvinenkoIra, 2019-10-21:
@JackLivio thank you for your response. I agree that this issue requires proposal and we from the Luxoft side can prepare it.

LitvinenkoIra, 2019-10-31:
Closed since Luxoft will prepare a proposal for this issue.

JackLivio, 2021-06-07:
Reopening since a proposal has been entered by Livio and is in review @LitvinenkoIra

jordynmackool, 2021-09-13:
The proposal mentioned in this issue was accepted (https://github.com/smartdevicelink/sdl_core/issues/3715)

iCollin, 2022-02-11:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3853

closed 2022-02-11 by iCollin