smartdevicelink/sdl_core #3077: SDL does not apply versioning for complex custom vehicle data

YarikMamykin, 2019-10-15:
Bug Report

SDL does not apply versioning for complex custom vehicle data

Preconditions

1. SDL and HMI are started with App 1 (veersion 3.0) and App 2 (version 5.9)
2. Policy table update includes next data:

```
{
  name = "custom_vd_item_struct",
  type = "Struct",
  key = "OEM_REF_STRUCT",
  mandatory = false,
  params = {
    {
      name = "struct_element_6_struct",
      type = "Struct",
      key = "OEM_REF_STRUCT_6_STRUCT",
      mandatory = false,
      params = {
        {
          name = "substruct_element",
          type = "Boolean",
          key = "OEM_REF_SUB_STRUCT",
          mandatory = false,
          since = "5.1"
        }
        {
          name = "substruct_element",
          type = "Integer",
          key = "OEM_REF_SUB_STRUCT",
          mandatory = false,
          since = "1.0",
          until = "5.0"
        }
      }
    }
  }
}
```

Reproduction Steps

App 1 requests GetVehicleData { custom_vd_item_struct  = true }

HMI responds to SDL with next data:
```
    {
      "id" : 39,
      "jsonrpc" : "2.0",
      "result" : 
      {
        "OEM_REF_STRUCT" : 
        {
          "OEM_REF_STRUCT_6_STRUCT" : 
          {
            "OEM_REF_SUB_STRUCT_2_BOOL" : 130
          }
        },
        "code" : 0,
        "method" : "VehicleInfo.GetVehicleData"
      }
    }
```

Expected Behavior
SDL processed the request successfully

Observed Behavior
SDL processes the request and respond with GENERIC_ERROR

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: []
* Testing Against: ATF test script

##### Test Case, Sample Code, and / or Example App
[]

jacobkeeler, 2019-10-17:
@YarikMamykin `OEM_REF_SUB_STRUCT_2_BOOL` does not match any of the keys in the provided in the policy table. I would imagine that the two versions of the parameter would have different keys, such as `OEM_REF_SUB_STRUCT_INT` and `OEM_REF_SUB_STRUCT_BOOL`?

I also don't think this makes much sense. Versioning information should not be included in custom data, so I would have to guess that this is meant to represent a future RPC spec item, correct?

In this case, it seems to me that the HMI would be sending a boolean value in `OEM_REF_SUB_STRUCT_2_BOOL`, which means that this issue wouldn't happen

mkorniichuk, 2019-10-18:
@jacobkeeler  actually there is a typo: there should be `OEM_REF_SUB_STRUCT` instead of `OEM_REF_SUB_STRUCT_2_BOOL`.
And since custom data indeed should not contain versioning data, the issue is similar to #3084 and so, not actual for now.

closed 2019-10-18 by mkorniichuk