smartdevicelink/sdl_core #3079: Core dump by ignition off after removing of custom vehicle data during PTU

GetmanetsIrina, 2019-10-16:
### Bug Report
Core dump by ignition off after removing of custom vehicle data during PTU

##### Reproduction Steps
1. App is registered and activated
2. custom_data_1 presets in DB
3. PTU is performed with removing custom_data_1
4. Ignition off is performed

##### Expected Behavior
SDL shuts down successful

##### Observed Behavior
Core dump

##### OS & Version Information
* OS/Version: [The OS/Version you were running when the bug occurred]
* SDL Core Version: [The version of SDL Core you were using when the bug occurred]
* Testing Against: [What you tested with to observe this behavior. Proxy, App, HMI etc.]

##### Test Case, Sample Code, and / or Example App
[Paste a link to a PR, gist, or other code that exemplifies this behavior]

GetmanetsIrina, 2019-10-16:
@theresalech, Please triage this issue.
Luxoft team is going to prepare a fix.

dboltovskyi, 2019-10-22:
@theresalech Please notice fix for core and updates for scripts are ready for review.

ShobhitAd, 2019-10-23:
Closing with merge of #3081

PR #3081 (2019-10-23): Fix VD items subscriptions after PTU
Fixes #3079  and #3080 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered with ATF scripts

### Summary
There was found an issue that SDL still kept subscriptions for custom vehicle data items even
for case when they were removed during PTU. By that reason, SDL was crashing with DCHECK on
attempt to unsubscribe from non existing custom data.

There was added missing logic of internal unsubscribe in case  some custom vehicle data was removed by policies.

### CLA
- [x] I have signed [the CLA](https://docs [... cut]
closed 2019-10-23 by ShobhitAd