smartdevicelink/sdl_core #3080: HMI does not send the VehicleInfo.UnsubscribeVehicleData to HMI in case subscribed custom data is removed during PTU

GetmanetsIrina, 2019-10-16:
### Bug Report
HMI does not send the `VehicleInfo.UnsubscribeVehicleData` to HMI in case subscribed custom data is removed during PTU

##### Reproduction Steps
1. App is registered and acrivated
2. App is subscribed for custom data data_1
3. PTU is triggered
4. PTU is performed successful with removing of custom data data_1 from schema_items

##### Expected Behavior
SDL removes the internal subscription for data_1 and sends `VI.UnsubscribeVehicleData(data_1)` to HMI

##### Observed Behavior
SDL does not send the `VI.UnsubscribeVehicleData(data_1)` to HMI by data removing

##### OS & Version Information
* OS/Version:Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/releases/tag/6.0.0_RC ( 2b09ea3 )
* ATF: [develop](https://github.com/smartdevicelink/sdl_atf/tree/develop)
* Script: TBD

##### Logs
[2019-10-16_14-35-13.zip](https://github.com/smartdevicelink/sdl_core/files/3734094/2019-10-16_14-35-13.zip)

theresalech, 2019-10-16:
@GetmanetsIrina please advise when the fix is ready for review.

dboltovskyi, 2019-10-22:
@theresalech Please notice fix for core and updates for scripts are ready for review.

ShobhitAd, 2019-10-23:
Closing with merge of #3081

closed 2019-10-23 by ShobhitAd