smartdevicelink/sdl_core #3084: HMI API does not support versioning

mkorniichuk, 2019-10-16:
### Bug Report
Currently, SDL does not support proper versioning since it's not designed to be applicable to HMI.

There are `since` and `until` attributes in MOBILE_API which allow validating messages according to certain rules.
Since HMI_API do not support these attributes, SDL cannot handle several API versions for HMI.
Currently, SDL validates HMI messages only against the latest version of API.

Due to this fact, for now, the proper functionality for mobile application, that uses an older API version, cannot be guaranteed.

##### Reproduction Steps
Preconditions:

MOBILE_API:
```
<function name="GetVehicleData" messagetype="request">
    <param name="VehicleDataParam" type="Integer" since="1.0" until="5.0">
    </param>
    <param name="VehicleDataParam" type="String" since="5.1">
    </param>
</function>
```

HMI_API:
```
<function name="GetVehicleData" messagetype="request">
    <param name="VehicleDataParam" type="String">
    </param>
</function>
```

1. GetVehicleData request from the mobile application with old API version:

Mobile request (using API version 4.5.1):
```
    "GetVehicleData":
    {
        VehicleDataParam = true
    }
```

#### HMI response (with Integer):
Despite message seem to be valid by MOBILE_API, but for SDL such message is invalid (by HMI_API) and will be rejected.

```
    {
        "result": {
            "code": 0,
            "method": "VehicleInfo.GetVehicleData",
            "VehicleDataParam": 1
        }
    }
```

##### Observed Behavior
SDL rejects message from HMI since it is not valid against HMI_API.

##### Expected Behavior
Successful processing of HMI response.

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: develop
* Testing Against: ATF test scripts

jacobkeeler, 2019-10-16:
@mkorniichuk I don't believe that the HMI API should be versioned as it stands. If this is desired, it should be a new proposal, as it puts a lot more work on the HMI. The issue being proposed is also not quite possible, as the HMI would send a String value in `VehicleDataParam`, not `1`, since it should match the HMI spec.
The proper solution would be for Core to translate this value for older apps:

the HMI Response would be:
```
    {
        "result": {
            "code": 0,
            "method": "VehicleInfo.GetVehicleData",
            "VehicleDataParam": "1"
        }
    }
```
and the Mobile response would be translated:
```
"GetVehicleDataResponse": {
  "resultCode": "SUCCESS",
  "success": true,
  "VehicleDataParam": 1
}
```

Beyond all of this, I don't believe that it is a good practice to ever change the `type` of a parameter, so I don't think this exact situation would arise anyway.

mkorniichuk, 2019-10-17:
@jacobkeeler thanks for your feedback

I agree the issue requires enough changes for a new proposal.

The data conversion within SDL does not seem to be the best solution from my point of view. Moreover, the type is not the only attribute causing the issue.
For example, if there are changes in `array` or `mandatory` attributes for a newer version of API, it's going to be really problematic to support old mobile applications.

jacobkeeler, 2019-10-17:
@mkorniichuk How do you propose that we handle notifications then?

If `VI.OnVehicleData` is sent by the HMI, it has no idea what applications this could be going to. The only solution in this case is for Core to translate the message.

I believe that there should be significant debate over changing these sorts of attributes, as discretion will always be needed when handling backward-compatibility with these. I don't believe that HMI API versioning works as a general solution. If you want an example of how an array can be handled, you can look at #2959 

Regardless, we agree that this would need a proposal, so I will close this issue for now. A new issue will be created if a proposal is accepted for it in the future.

closed 2019-10-17 by jacobkeeler