smartdevicelink/sdl_core #3096: Application does not retrieve subscriptions to app service after resumption

YarikMamykin, 2019-10-22:
### Bug Report
Application does not retrieve subscriptions to app service after resumption.

##### Reproduction Steps
1. Register application
2. Subscribe application to app service(s)
3. Perform unexpected disconnect
4. Perform resumption

##### Expected Behavior
After resumption application retrieves all its subscriptions to app service(s)

##### Observed Behavior
After resumption application **does not** retrieves all its subscriptions to app service(s)

##### OS & Version Information
* OS/Version: [Ubuntu 18.04]
* SDL Core Version: [https://github.com/smartdevicelink/sdl_core/tree/develop](url)
* Testing Against: ATF test scripts + UT

##### Test Case, Sample Code, and / or Example App
[TBD]

dboltovskyi, 2019-10-25:
@theresalech Please triage this issue.
Fixes in `core` and `scripts` are ready for review.

ShobhitAd, 2019-10-28:
Closing with the merge of https://github.com/smartdevicelink/sdl_core/pull/3097

PR #3097 (2019-10-28): Fix app service extension resumption data processing
Fixes #3096 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
ATF Regression testing + UT

### Summary
In case of resumption app services subscription data for particular application lost every time due to incorrect behaviour of AppServiceAppExtension::ProcessResumption method logic.
With fix resumption data is treated correctly.

### CLA
- [X] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2019-10-28 by ShobhitAd