smartdevicelink/sdl_core #3098: ExpectPolicyTableReset unit test is unstable

dboltovskyi, 2019-10-22:
### Bug Report
Sometimes (~30%) `ResetPolicyTable_WithPreloadedFile_ExpectPolicyTableReset` test which is in `policy_handler_test` may fail.

##### Reproduction Steps
1. Build SDL with UT
2. Run described unit test
3. Check results

##### Expected Behavior
Test is passed

##### Observed Behavior
Test is failed with message:
```
11:40:58 Uninteresting mock function call - taking default action specified at:
11:40:58 /src/components/application_manager/test/policy_handler_test.cc:176:
11:40:58     Function call: event_dispatcher()
11:40:58           Returns: @0x326e938 8-byte object <F8-A0 E9-02 00-00 00-00>
11:40:58 Stack trace:
11:40:58 /src/components/application_manager/test/policy_handler_test.cc:370: Failure
11:40:58 Value of: policy_handler_.ResetPolicyTable()
11:40:58   Actual: false
11:40:58 Expected: true
```

##### OS & Version Information
* OS/Version: Ubuntu 16.04 amd64
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (1d3a401)
* Testing Against: N/A

##### Test Case, Sample Code, and / or Example App
  - Build log: [build.log.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3757646/build.log.tar.gz)

dboltovskyi, 2019-10-25:
@theresalech Please triage this issue.
Fix in `core` is ready for review.

jacobkeeler, 2019-10-28:
Closing with the merge of #3099

PR #3099 (2019-10-28): Synchronize backup and reset policy functions
Fixes #3098 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered with unit tests

### Summary
This is an issue specific for `EXTERNAL_PROPRIETARY` flow only. This issue is exposed by `PolicyHandlerTest` unit test: `ResetPolicyTable_WithPreloadedFile_ExpectPolicyTableReset`
which sometimes is failing.

The main problem was that database reset(drop) and backup can be done simultaneously in different threads which were not synchronized. In that case, database drop often may be failed because another  [... cut]
closed 2019-10-28 by jacobkeeler