smartdevicelink/sdl_core #3108: SDL crashes after all apps exited

AByzhynar, 2019-10-31:
### Bug Report
SDL crashes after all apps exited.

##### Precondition
1. HMI and SDL are started
2. 2 Apps are registered and activated 

##### Reproduction Steps
Unknown. 
SDL core crash happens during manual testing of RC Multiple Modules functionality.
2 app instances were involved.
After exiting of the last app (via Menu->Exit) - in 2-3 seconds SDL crashed.

##### Occurence
ONCE

##### Expected Behavior
SDL works normally without crashes

##### Observed Behavior
SDL crashed 

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/releases/tag/6.0.0_RC2 ( 4d283e )
* HMI: https://github.com/smartdevicelink/sdl_hmi/tree/develop ( bd53f1 ) 
* SPT: SyncProxyTester ( 20191028 Nightly-Android, Proxy Lib Ver 4.9.1 ) 

Core GDB Backtrace:
[gdb.txt](https://github.com/smartdevicelink/sdl_core/files/3793440/gdb.txt)

AByzhynar, 2019-10-31:
After issue back trace analysis - the crash reason is clear. During OnSystemCapabilityUpdated notification - app was not possible to extract required extension. Thus DCHECK (ASSERT) happens. But it is unclear how an extension could be absent.

dboltovskyi, 2019-10-31:
Since the issue is happened only once it doesn't know whether it regression one or not.
Additional analysis is required.

dboltovskyi, 2019-10-31:
@theresalech Please triage this issue.

theresalech, 2019-10-31:
@dboltovskyi please let us know once your additional analysis is complete, and we'll review.  Thanks!

LitvinenkoIra, 2019-11-01:
@AByzhynar it seems like the same crash was reproduces on my side. Please check the attached logs 
[2019-11-01_15-40-30.zip](https://github.com/smartdevicelink/sdl_core/files/3798160/2019-11-01_15-40-30.zip)

theresalech, 2019-11-06:
@LitvinenkoIra @AByzhynar @IGapchuk are you still investigating this issue? Do you have plans to submit a fix for Livio review?

IGapchuk, 2019-11-06:
@theresalech We have finished the investigation and prepared a fix for it and the fix now under testing. We'll provide PR with the fix as quickly as possible after successful testing.

JackLivio, 2019-11-21:
Fixed with merge of #3115

PR #3115 (2019-11-21): Fix SDL crash during application registration
Fixes #3108 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
There is an issue when SDL crashes during registration second application in the time when already exists a registered other one application.

In the current implementation ApplicationManager during application registering by the `ApplicationManager::RegisterApplication` method creates a new instance of the application and adds it to the collection of all registered applications. This method calls in the  RegisterAppInterfaceRequest, but all need [... cut]
closed 2019-11-21 by JackLivio