smartdevicelink/sdl_core #3109: SDL Crash - App registration response

AByzhynar, 2019-10-31:
### Bug Report
SDL crashed during app registration

##### Reproduction Steps
1. Make sure SDL is built in `EXTERNAL_PROPRIETARY` mode
2. Run SDL, HMI, connect Mobile
3. Register new app

##### Expected Behavior
App registered successfully

##### Observed Behavior
SDL crashed during sending of RAI response

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/releases/tag/6.0.0_RC2 ( 4d283e )
* HMI: https://github.com/smartdevicelink/sdl_hmi/tree/develop ( bd53f1 ) 
* SPT: SyncProxyTester ( 20191028 Nightly-Android, Proxy Lib Ver 4.9.1 ) 

##### Attachments

dboltovskyi, 2019-10-31:
Since the issue is happened only once it doesn't know whether it regression one or not.
Additional analysis is required.

dboltovskyi, 2019-10-31:
@theresalech Please triage this issue

theresalech, 2019-10-31:
@dboltovskyi please let us know once your additional analysis is complete, and we'll review.  Thanks!

iCollin, 2019-11-05:
@AByzhynar @dboltovskyi 

I have run [144_ATF_PTU_validation_failure](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Policies/Policy_Table_Update/144_ATF_PTU_validation_failure.lua) over 250 times now without seeing the segmentation fault.

Deleting a SmartObject of type Map is the cause, but diagnosing via static analysis will be difficult.

Have you been able to reproduce this issue?

dboltovskyi, 2019-11-05:
@iCollin Unfortunately we caught this crash only once with script mentioned. 
Though potential fix is ready and published in [fix/fix_undefined_behavior_after_so_cleanup](https://github.com/smartdevicelink/sdl_core/tree/fix/fix_undefined_behavior_after_so_cleanup) branch.

iCollin, 2019-11-06:
@dboltovskyi While I think the changes in [fix/fix_undefined_behavior_after_so_cleanup](https://github.com/smartdevicelink/sdl_core/tree/fix/fix_undefined_behavior_after_so_cleanup) cannot hurt, I do not think they will fix the issue. In [gdb (8).txt](https://github.com/smartdevicelink/sdl_core/files/3793571/gdb.8.txt), the crash seems to arise from a bad pointer in the `ISchemaItemPtr`.

dboltovskyi, 2019-11-06:
@AKalinich-Luxoft Could you add some details please?

AKalinich-Luxoft, 2019-11-06:
@iCollin that's why it's a potential fix as we can not be 100% sure that issue may be fixed by these changes. However, one of the reasons of invalid pointer in `ISchemaItemPtr` might be a double release of the same memory address by raw pointer in `cleanup_data` which is in the 21st frame of the stack trace. So these changes prevent possible double deletion and if it was the root cause - it solves the issue itself.

iCollin, 2019-11-06:
@AKalinich-Luxoft As far as I can tell, `cleanup_data` only deals with `SmartObject::m_data` and `SmartObject::m_type`. `~SmartObject` will go on to destroy `m_schema` which contains the `ISchemaItemPtr` after `cleanup_data` has returned. I don't understand how an invalid `ISchemaItemPtr` would be caused by code within `cleanup_data`.

AKalinich-Luxoft, 2019-11-08:
@iCollin yes, `cleanup_data` deals with `SmartObject::m_data` and when smart object is represented as a map, `m_data` is pointing to `map_value` which is a map of strings in a key and another `SmartObject` as a value. 
So my theory is that if `cleanup_data` was called twice for the same SmartObject and this smart object is holding a pointer to `map_value` then this map might be deleted twice and `~SmartObject` might be called twice for some smart object in a map. As a result, `m_schema` will be deallocated twice and we may see a backtrace like in the GDB file attached. But again, it's just a theory that's why fix is potential. For now I can't see any ways how to reproduce that.

iCollin, 2019-11-08:
@AKalinich-Luxoft thank you for explaining. I previously wasn't considering that `cleanup_data` is deleting children `SmartObject`s. In the case you describe, the second delete call would seg fault immediately as the memory pointed to by `map_value` has been deallocated, correct? I do not think there would be a second call to `~SmartObject`. Your code is a good addition, but I still do not understand how it could solve the issue described here.

AKalinich-Luxoft, 2019-11-08:
@iCollin correct, looks like we will get seg fault first. Well, in that case I agree this fix is not solving the issue completely. My another theory is that we have two separate smart objects and each of them is holding `CSmartSchema` instance with the shared pointer to the same schema and both shared pointers are independent (reference_count = 1 for both), so when one of smart objects is destroyed, its smart schema destroys schema instance by shared pointer. When the second smart object is destroyed, its smart schema is trying to destroy the same schema instance (which was already destroyed) by its own shared pointer. I mean that in some place in SDL code there was something like `make_shared` twice for the same object in memory. To find the root cause, we should check all places of initialization of `CSmartSchema` and find such place.
I suspect it might be related to changes brought by RGNSD feature as in that feature `ISchemaItemPtr` was actively used.

theresalech, 2019-11-12:
@AKalinich-Luxoft thank you for the update.  Please let us know once you are able to diagnose root cause, and/or if you're able to reproduce this issue again.

AKalinich-Luxoft, 2019-11-15:
Something similar may be reproduced within the unit test like this:
```
TEST(invalid_schema, simple_test2) {
  SmartObject obj;
  ISchemaItemPtr item = CAlwaysTrueSchemaItem::create();
  CSmartSchema schema(item);

  obj.setSchema(schema);
  delete item.get();
}
```

AKalinich-Luxoft, 2019-11-15:
@iCollin I think the root cause here is related to HMICapabilities class. So when one thread is running RAI request, at some places it copies smart object structures by raw pointer. If at the same time another thread is trying to set the same structure, it destroys a previous one by the same raw pointer, so in some rare cases we may have a corrupted copy of `SmartObject` (or even segfault). A corrupted copy of the `SmartObject` may contain a `CSmartSchema` with the shared pointer to initial schema item which was destroyed by another thread. After that, when corrupted smart object is going to be released, SDL crashes as shared pointer is trying to release the same schema item the second time. So I updated all places of usage of raw pointers in the scope of RAI request. I can also update all other places, but it would be out of this issue scope.
Please check for my last commit in the linked PR.

AKalinich-Luxoft, 2019-11-25:
@theresalech PR is ready for Livio team review

jordynmackool, 2019-11-25:
@AKalinich-Luxoft thanks, we will review and let you know if we have any questions.

iCollin, 2020-01-08:
Closed via #3116

PR #3116 (2020-01-07): Fix undefined behavior after changing HMICapabilities smart object field
Fixes #3109 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by ATF tests

### Summary
After `cleanup_data()` call, delete is used to deallocate memory by pointer, however pointer still hold a value of address of deallocated memory. By that reason SO still have an ability to access that data what might cause UB including core crash in the random places.

This function has been updated to set pointers to NULL as well as smart object type to prevent accident access to  [... cut]
closed 2020-01-08 by iCollin