smartdevicelink/sdl_core #3110: SDL Crash - SegFault during execution of transport_manager unit tests

AByzhynar, 2019-10-31:
### Bug Report
SDL crashed during app registration

##### Reproduction Steps
1. Build SDL Core with "BUILD_TESTS=On" option
2. Run Transport Manager Unit tests

##### Expected Behavior
All unit tests are PASSED

##### Observed Behavior
SDL crashed during transport manager unit tests

##### Occurrence:
Sometimes

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/releases/tag/6.0.0_RC2 ( 4d283e )
* HMI: https://github.com/smartdevicelink/sdl_hmi/tree/develop ( bd53f1 ) 
* SPT: SyncProxyTester ( 20191028 Nightly-Android, Proxy Lib Ver 4.9.1 ) 

##### Attachments
[build_log.txt.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3793577/build_log.txt.tar.gz)
[coredump.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3793578/coredump.tar.gz)

AByzhynar, 2019-10-31:
After issue analysis - it appears that the problem is in the SDL code but not in unit tests. The problem is that after TM destruction - one thread which has created usb devices (as well as used file descriptors for them) is in lock state. Thus this thread does not free acquired resources. Threfore when the tests run multiple times - OS has no free dile descriptors - as result we got core crash.

theresalech, 2019-10-31:
@dboltovskyi please let us know once your additional analysis is complete, and we'll review.  Thanks!

theresalech, 2019-11-06:
@AStasiuk @dboltovskyi are you still investigating this issue?  Do you have plans to submit a fix for Livio review?

dboltovskyi, 2019-11-06:
@theresalech Yes, we're testing a potential fix now. We will notify PM once PR is ready for review.

dboltovskyi, 2019-11-08:
@theresalech Please notice fix in https://github.com/smartdevicelink/sdl_core/pull/3121 is ready for PM review.

Please take into account this fix reveals two other issues:
 - https://github.com/smartdevicelink/sdl_core/issues/2465
 - https://github.com/smartdevicelink/sdl_core/issues/3118

Fixes for them going to be provided in separate PRs

theresalech, 2019-11-19:
Closed via #3121

PR #3121 (2019-11-18): Rework usb handler delegate completion
Fixes #3110

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Tested by existing unit tests

### Summary
The problem was in the Thread() function of USB handler delegate. It used blocking call `libusb_handle_events_completed()`. So when USB handler dtor was called - delegate was destroyed in force way without free of allocated resources. Therefore it leads to file descriptors leak and as a result core crash when the number of available file descriptors exceeded OS maximum per user.

### CLA
- [x] I have signed [... cut]
closed 2019-11-19 by theresalech