smartdevicelink/sdl_core #3111: SDL switches applications into incorrect HMILevel after IGNOTION_OFF/IGNITION_ON cycle

IGapchuk, 2019-11-01:
### Bug Report
There is an issue when SDL switches applications into incorrect HMILevel after IGNOTION_OF/IGNITION_ON cycle.

##### Reproduction Steps
1. SDL and WebHMI are successfully started.
2. Register and activate App1 (SDL switches App1 into HMILevel::FULL).
3. Register and activate App2 (SDL switches App2 into HMILevel::FULL and switches App1 into HMILevel::BACKGROUND).
4. Send "BasicCommunication.OnExitAllApplications","params":{"reason":"IGNITION_OFF"} from WebHMI.
5. SDL is stopped successfully.
6. Start SDL (the new ignition cycle).
7. SDL and WebHMI are successfully started.
8. Register and activate the App1 and App2 without resumption.

##### Expected Behavior
1. Register and activate App1 (SDL switches App1 into HMILevel::FULL).
2. Register and activate App2 (SDL switches App2 into HMILevel::FULL and switches App1 into HMILevel::BACKGROUND).

##### Observed Behavior
[1. Register and activate App1 (SDL switches App1 into HMILevel::FULL).
2. Register and activate App2 (SDL switches App2 into HMILevel::NONE and switches App1 into HMILevel::LIMITED).

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: release/6.0.0 (4d283e625dfa09d166d572fa7bf6e98e91f05050)
* WebHMI: develop (https://github.com/smartdevicelink/sdl_hmi/commit/bd53f11339ee4d95a41caf5e5edf4ae0839d5957)

SDL log: 
[SmartDeviceLinkCore.log.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3797531/SmartDeviceLinkCore.log.tar.gz)
WebHMI log: 
[HMI_log.log.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3797533/HMI_log.log.tar.gz)

JackLivio, 2019-11-01:
@IGapchuk Thank you for noting this issue. Were both apps located on the same device?

IGapchuk, 2019-11-01:
@JackLivio Yes, both applications were located on the same device. We've retested the same test case from two devices and the result is the same. Appropriate logs are attached.
[InCorrectHMILevel_two_devices.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3798275/InCorrectHMILevel_two_devices.tar.gz)

dboltovskyi, 2019-11-03:
It seems issue was introduced with merge of https://github.com/smartdevicelink/sdl_core/pull/2980

dboltovskyi, 2019-11-03:
Appropriate ATF script is attached and mentioned in description

JackLivio, 2019-11-06:
This issue is not reproducible when I try it manually. I am using most recent master branches of SDL Core and SDL HMI.

After I restart Core and HMI, the first app that registers is not put into full. When I register the second app, the second app is resumed into full. I am observing the correct behavior.

dboltovskyi, 2019-11-06:
@JackLivio Please notice after Ignition cycle 1st app needs to be activated and deactivated withing 3 sec timeout
This is default duration before SDL starts resumption of apps' HMI levels (in .INI this is `ApplicationResumingTimeout` parameter)

theresalech, 2019-11-14:
Closed via #3117

PR #3117 (2019-11-13): Check for conflicting app resumption level
Fixes #3111 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Use ATF script GH-3111.lua supplied in issue.

### Summary
When the resume controllers timer fires to resume an app, there is no check to see if the saved hmi level conflicts with an app that is already in a limited or full state. This fix looks for that case and returns false for RestoreAppHMIState. This will mark the resumption as failed and set the app is_resuming state to false. This allows an app to later be activated via SDL.ActivateApp.
 [... cut]
closed 2019-11-14 by theresalech