smartdevicelink/sdl_core #3112: SDL sends unsucces general response to mobile app in case choiceID was selected on UI via PerformInteraction(BOTH)

GetmanetsIrina, 2019-11-05:
### Bug Report
SDL sends unsucces general response to mobile app in case choiceID was selected on UI via PerformInteraction(BOTH)

##### Reproduction Steps
1. HMI and SDL are started
2. App is registered and activated
3. CreateInteractionChoiceSet is created
4. App requests PerformInteraction(BOTH)
5. SDL requests UI.PI and VD.PI
6. HMI shows VR portion
7. Abort VR portion/Wait timeout for VR.PI
8. HMI sends VR.PI(ABORTED)/ VD.PI(TIMED_OUT)
9. HMI shows UI portion
10. Pick choice on UI
11. HMI sends UI.PI(SUCCESS, choiceID)

##### Expected Behavior
SDL sends PerformInteraction(SUCCESS, true, choiceID) response to mobile app.
[image]

##### Observed Behavior
SDL sends PerformInteraction(ABORT/TIMED_OUT, false, choiceID) response to mobile app

##### OS & Version Information
* OS/Version:  Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/release/6.0.0 (tag RC_2)
https://github.com/smartdevicelink/sdl_core/tree/develop
* HMI: https://github.com/smartdevicelink/sdl_hmi/tree/develop (bd53f11339ee4d95a41caf5e5edf4ae0839d5957) 
* SPT: https://rink.hockeyapp.net/apps/dbc7db5274027027262185753824e34d
Version 20191028-Nightly-Android (1034) 
* Script: TBD

##### Logs
[SmartDeviceLinkCore.zip](https://github.com/smartdevicelink/sdl_core/files/3808688/SmartDeviceLinkCore.zip)

iCollin, 2019-11-12:
Related to https://github.com/smartdevicelink/sdl_core/issues/2990

YarikMamykin, 2019-11-19:
@theresalech Please triage this issue and notice linked fix is ready for review.

theresalech, 2019-11-19:
@YarikMamykin both linked PRs (https://github.com/smartdevicelink/sdl_core/pull/3122, https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2293) have `This PR is **not ready** for review.` in their descriptions.  Please tag me and @jordynmackool on those PRs when they are confirmed as ready for review and descriptions have been updated.

YarikMamykin, 2019-11-25:
@theresalech please notice that PRs #3122 and smartdevicelink/sdl_atf_test_scripts#2293 are ready for review.

iCollin, 2020-01-31:
Closed via #3122

closed 2020-01-31 by iCollin