smartdevicelink/sdl_core #3114: SDL sends StartServiceAck(encryption: true) in handshake only mode

AKalinich-Luxoft, 2019-11-06:
### Bug Report
SDL should consider flag `NonEncryptedAfterHandshake` and work according to the following requirement:

> In case
> "NonEncryptedAfterHandshake" param has `<value_1>` and app with `<value_1>`-transport successfully connects and opens RPC service (RPC = 7 service) and this app sends StartService (`<any_serviceType>`, encryption=true) to SDL and TLS handshake is successfull
> 
> SDL must:
> respond StartService (`<any_serviceType>`, encryption=false) to this mobile app

##### Reproduction Steps
1. In SDL ini file, set `NonEncryptedAfterHandshake` flag in `Security Manager` section to `WIFI`
2. Start SDL, HMI
3. On Android device, run Antelope app
4. Set `IP address`, set `Security Type` to `HS_ONLY`, click Ok, restart Antelope
5. Antelope is registered on SDL and displayed on HMI
6. On HMI, activate Antelope and click OK for a start service request

##### Expected Behavior
1. Antelope sends StartService(protection: ON) for a video service
2. SDL performs SSL handshake and finishes it succesfully
3. SDL sends StartServiceAck(protection: OFF) for a video service
4. Waze starts streaming in unprotected mode

##### Observed Behavior
1. Antelope sends StartService(protection: ON) for a video service
2. SDL performs SSL handshake and finishes it succesfully
3. SDL sends StartServiceAck(protection: ON) for a video service
4. Waze starts streaming in protected mode

##### OS & Version Information
* OS/Version: Ubuntu 16.04 LTS
* SDL Core Version: 6.0.0_RC2 (4d283e625dfa09d166d572fa7bf6e98e91f05050)
* Testing Against: Antelope 5.8.4

dboltovskyi, 2019-11-06:
@theresalech Please triage this issue. 
Currently one of the OEMs is experiencing video data distortion without this issue been fixed.
We're not sure is this could be fixed as a Bug or it requires a Proposal. 
Please suggest.
Thank you.

JackLivio, 2019-11-06:
`NonEncryptedAfterHandshake` does not exist in the core project from what i can tell.

JackLivio, 2019-11-06:
If this is a parameter you are asking to add to the project then a proposal will be required.

TinaKleczka, 2019-11-12:
@theresalech 
I am hoping the PM may be able to provide me with some clarity on why this PR was closed so quickly.  I know Joel added this feature and was working with Waze directly.  Looks like we introduced the encryption methods in Aug. 2016 9v4.2.0) .  
 
We have proxy code which supports the missing Encryption parameter on iOS. This parameter is supported in SYNC 3.
We believe the missing encryption parameter may have just been overlooked during development in the Open and Android. 
Could you confirm?
There is a risk of this parameter not being developed in the Open and in the mobile Proxy if other OEM’s would like to use Waze.  
 
Jack closed the Bug pretty quickly and suggested we needed a proposal to add the parameters.  We believe this would be a bug and are hoping you can provide some clarity as to why this would need a new proposal?  
 
We are not able to find any documentation regarding this feature, but we have noticed the option for the handshake was only implemented in iOS.

theresalech, 2019-11-12:
Re-opening issue to investigate further.

iOS References, as provided by Tina via email:

- https://github.com/smartdevicelink/sdl_ios/blame/4.2.0/SmartDeviceLink/SDLStreamingMediaManager.h#L30
- https://github.com/smartdevicelink/sdl_ios/blob/master/SmartDeviceLink/SDLStreamingMediaManagerConstants.h#L20

theresalech, 2019-11-14:
@TinaKleczka @AKalinich-Luxoft we originally closed this issue after determining the parameter did not exist in Core, and before the iOS references were cited.  However, we've now looked into the flag on iOS and found the following:

1. The `authenticateOnly` flag on iOS currently has the same behavior as the `authenticateAndEncrypt` flag. There is no difference between them.
2. When it was originally created, the `authenticateOnly` flag did have different behavior. It would _send_ non-encrypted data after the handshake no matter what the head unit indicated should be done. This is, in our current opinion, bad behavior.
3. Based on the original and current behavior, this flag doesn't relate to the current issue in any way other than a similarity of naming.
4. The current issue under discussion takes advantage of behavior in the iOS app library (to be confirmed for Java Suite library) that change encryption based on the latest `StartServiceAck` received. Discussion around if this behavior should be solidified into the INI file of Core requires additional discussion that should take place in the proposal process. This seems to be a feature, not a bug.
5. This feature should be service independent since we support RPC service encryption, however, it's taking advantage of behavior that exists only in the iOS video / audio managers. Additional implementation would then need to take place on iOS and Java app libraries.

We'd therefore recommend submitting a proposal to implement this feature, as we do believe it provides value to the project, but will require documentation for how the feature should be implemented, with respect to existing implementations (ie SYNC 3).  This issue posts a requirement that has not yet been adopted by the SDLC.

Please let us know if you have any questions about this recommendation.  We'll otherwise plan to close this issue.

theresalech, 2019-11-19:
Closing per agreement to comment above.

closed 2019-11-19 by theresalech