smartdevicelink/sdl_core #3119: App gets "REJECTED" instead of "UNSUPPORTED_RESOURCE" to ChangeRegistration with not supported language

GetmanetsIrina, 2019-11-07:
### Bug Report
App gets "REJECTED" instead of "UNSUPPORTED_RESOURCE" to ChangeRegistration with not supported language

##### Reproduction Steps
1. HMI does not support German language
2. HMI and SDL are started
3. HMI has set English language for UI and TTS, VR
4. App is registered with EN language
5. App requests ChangeRegistration(DE-DE) for UI/TTS,VR

##### Expected Behavior
SDL sends response ChangeRegistration(UNSUPPORTED_RESOURCE) to mobile app

##### Observed Behavior
SDL sends response ChangeRegistration(REJECTED) to mobile app

##### OS & Version Information
* OS/Version:  Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/release/6.0.0 (tag RC_2)
https://github.com/smartdevicelink/sdl_core/tree/develop
* HMI: https://github.com/smartdevicelink/sdl_hmi/tree/develop (bd53f11339ee4d95a41caf5e5edf4ae0839d5957) 
* SPT: https://rink.hockeyapp.net/apps/dbc7db5274027027262185753824e34d
Version 20191028-Nightly-Android (1037) 
* Script: TBD

##### Logs
TBD

E-SAITO-TMC, 2020-05-13:
@theresalech -san, We were able to reproduce this issue with Core 6.0.1. This is a medium priority issue for Toyota. Could you please include this in the Core 6.2 release planning?

theresalech, 2020-05-21:
@GetmanetsIrina can you please advise if you are able to reproduce this issue in the latest Core release?

theresalech, 2020-06-11:
Hi @dboltovskyi, thank you for the additional information.  Can you please advise if Luxoft will be submitting a PR to resolve this issue?

dboltovskyi, 2020-06-11:
@theresalech Please notice we're not planning to create the fix in scope of 6.2, though will try to do as much as possible

ShobhitAd, 2020-06-30:
Closed via #3427

PR #3427 (2020-06-30): Fix App gets REJECTED instead of UNSUPPORTED_RESOURCE to Cha…
Fixes #3119 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by manual testing

### Summary
When `ChangeRegistration` request with not supported language, SDL uses `UNSUPPORTED_RESOURCE` as `ChangeRegistration` response `result.class`.
[image]
use `UNSUPPORTED_RESOURCE` replace `REJECTED` as `ChangeRegistration` response Result in `ChangeRegistrationRequest` class when receiving `ChangeRegistration` request with not supported language.

### CLA
- [x] I have signed [the CLA](htt [... cut]
closed 2020-06-30 by ShobhitAd