smartdevicelink/sdl_core #3123: There is no mechanism for managing the switching sequence(Start/StopStreaming) between two navigation applications.

zhouxin627, 2019-11-14:
### Bug Report
There is no mechanism for managing the switching sequence(Start/StopStreaming) between two navigation applications.

##### Reproduction Steps
1. Register 2 apps:
・App1: Waze for iOS
・App2: Sygic for iOS
2. Activate App1
3. App1 starts video streaming
4. Activate App2
5. App1 is not displayed in the HMI
    It runs continuously in the background.

##### Expected Behavior
1. Activate App1: SDL->App1.OnHMIStatus(FULL)
2. App1 starts Video streaming
3. Activate App2: SDL->App2.OnHMIStatus(FULL)
4. Deactivate App1: SDL->App1.OnHMIStatus(BACKGROUND)
5. Receive EndService request from App1:SDL<-App1.EndServiceCallback(11)
6. Send StopSteaming notification of App1:HMI<-SDL:OnVideoDataStreaming(False)
7. Receive StartService request from App2:SDL<-App2.StartServiceCallback(11)
8. Send StarSteaming notification of App2:HMI<-SDL:OnVideoDataStreaming(True)
9. App2 starts video streaming

##### Observed Behavior
If App1.EndServiceCallback(11) arrives later than App2.StartServiceCallback(11)，
HMI failed to create socket for the video streaming of the App2,
because the socket of App1 hasn't been disconnected.

##### OS & Version Information
* SDL Core Version: 5.1.3
* Testing Against: SDL Core 5.1.3 + HMI running on our development environment

##### Test Case, Sample Code, and / or Example App
Logs:[Log.zip](https://github.com/smartdevicelink/sdl_core/files/3844217/Log.zip)

##### Occurrence:
100%

##### Recovery Method:
None

E-SAITO-TMC, 2019-11-28:
This is one of the issues that Toyota reported at the 11/18 Roadmap workshop.

iCollin, 2020-01-31:
Hello @zhouxin627,

Can you clarify what you mean by "HMI failed to create socket for video streaming"? 
I am confused because the mobile app will send data to Core, which will be forwarded to mobile. 
Is the connection between HMI and Core being broken, or is the issue with the mobile app's streaming connection to core?

Can you also help me to understand the logs posted? I only see StartStream sent once.
As I understand the current streaming transition process is as follows: when app2 takes over streaming from app1, app1 will be put into the HMI level BACKGROUND and will stop streaming soon, but maybe not immediately. After a (default) delay of 3000ms, if app1 is still streaming, Core will forcibly unregister the app.

Thank you for your help!

zhouxin627, 2020-02-04:
> Can you clarify what you mean by "HMI failed to create socket for video streaming"?
> I am confused because the mobile app will send data to Core, which will be forwarded to mobile.
> Is the connection [...]

Hi, @iCollin Sorry, perhaps I didn't make myself clear.
It means when using a socket for VideoStreaming data transmission between HMI and SDLCore,
HMI as Socket Client and SDLCore as Socket Server.
When this issue occurs, HMI received StopStream notification then closed the socket connection.
This caused VideoStreaming data transmission was interrupted.

> Can you also help me to understand the logs posted? I only see StartStream sent once.
> As I understand the current streaming transition process is as follows: when app2 takes over streaming from [...]

Sorry, the attached log file is incomplete. I've updated it.

Step1. Start Waze app（App1）
```
Line 66717: 01-01 09:01:25.235 2172 3998 D MWSDL : [SDLNativeAggregator.cpp:1249@CtrlInvokeApp] SDLCtrl_InvokeApp : strAppId = 1333136227, option = 0
Line 73804: 01-01 09:01:25.787 2172 3890 D SDL : [application_manager_impl.cc:1949@OnServiceStartedCallback] ServiceType = 11. Session = 10003
Line 73893: 01-01 09:01:25.791 2172 3990 D MWSDL : [SDLWebsocketSession.cpp:279@Read] ######## Read<<:>Line 77218: 01-01 09:01:28.491 2172 3990 D MWSDL : [SDLWebsocketSession.cpp:279@Read] ######## Read<<:>Line 77867: 01-01 09:01:28.895 2172 3990 D MWSDL : [SDLWebsocketSession.cpp:279@Read] ######## Read<<:>
```

Step2. Start Sygic app（App2）
`Line 86118: 01-01 09:01:35.619 2172 3998 D MWSDL : [SDLNativeAggregator.cpp:1249@CtrlInvokeApp] SDLCtrl_InvokeApp : strAppId = 1797797493, option = 0`

Step3. Received StartVideoService of Sygic App, run 7～10 in the sequence digraph.
※VideoStreaming data transmission thread still working and will be reused, 
because Core hasn't received StopVideoService of Waze App.
```
Line 88192: 01-01 09:01:35.851 2172 3890 D SDL : [application_manager_impl.cc:1949@OnServiceStartedCallback] ServiceType = 11. Session = 100011
Line 88284: 01-01 09:01:35.851 2172 3990 D MWSDL : [SDLWebsocketSession.cpp:279@Read] ######## Read<<:>Sequence 10
Line 88451: 01-01 09:01:35.863 2172 3990 D MWSDL : [SDLWebsocketSession.cpp:279@Read] ######## Read<<:>Line 90923: 01-01 09:01:36.111 2172 3990 D MWSDL : [SDLWebsocketSession.cpp:279@Read] ######## Read<<:>
```

Step4.Received StopVideoService of Waze app, run 11～14 in the sequence digraph.
※SDLCore side: `appkey` of App2 is saved at this moment，so the judgment conditions were not met ，it did not close the socket connection.
※HMI side: Received StopStream and OnVideoStreamingData(false), but is unable to determine which app the notification came from，because the notification did not include AppID. 
So, it would close the socket connection.

```
Line 91038: 01-01 09:01:36.123 2172 3890 D SDL : [application_manager_impl.cc:1995@OnServiceEndedCallback] OnServiceEndedCallback for service 11 with reason 0 in session 0x10003
Line 91602: 01-01 09:01:36.159 2172 3990 D MWSDL : [SDLWebsocketSession.cpp:279@Read] ######## Read<<:>Sequence 14
Line 91608: 01-01 09:01:36.159 2172 3990 D MWSDL : [SDLWebsocketSession.cpp:279@Read] ######## Read<<:>
```
[image]

iCollin, 2020-02-04:
Hello @zhouxin627,

Thank you for your detailed explanation.

In your HMI implementation, would streaming continue without issue if the socket simply was not closed?
If that would not solve the issue, you can ignore my follow up question.

Otherwise, I am curious what causes the HMI to close the streaming socket. If `OnVideoStreamingData(false)` triggers the socket closing, it will be important to take a look at an issue that will be fixed in the next release of SDL Core: [SDL sends misplaced OnVideoDataStreaming notification](https://github.com/smartdevicelink/sdl_core/issues/3142), and [it's corresponding fix](https://github.com/smartdevicelink/sdl_core/pull/3154). With this fix in place, the `OnVideoStreamingData(false)` described in steps 11-14 will not be sent to the HMI.

zhouxin627, 2020-02-07:
> In your HMI implementation, would streaming continue without issue if the socket simply was not closed?
> If that would not solve the issue, you can ignore my follow up question.

Our HMI needs to disconnect and reconnect socket when switching NaviApp1 to NaviApp2, 
because the property from SetVideoConfig of NaviApp2 may be different. (e.g. video resolution, Codec type)
If we don't do that, NaviApp2 would not send SetVideoConfig.

> Otherwise, I am curious what causes the HMI to close the streaming socket. If OnVideoStreamingData(false) triggers the socket closing, it will be important to take a look at an issue that will be [...]

I think it is different from our case.
This PR can solve the issue of `OnVideoDataStreaming(false)` sending,
but it can't solve the need of reestablishing `VideoStreaming`.

iCollin, 2020-02-07:
Closed via #3217

PR #3217 (2020-02-07): Fix for No mechanism for managing the switching sequence(Start/StopSt…
Fixes #3123

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by Unit tests.

### Summary
**Reason**
1. Because switch between NaviApps. Only one app can be active at a time. Asynchronous notifications between apps may have timing issues.									
eg: App1.EndServiceCallback(11) arrives later than App2.StartServiceCallback(11)									
2. SDLCore does not effectively cover this case. Duplicate notifications affect the activated App.

**Solution**
1. When NaviApp2 activat [... cut]
closed 2020-02-07 by iCollin